RavenSystem/esp-homekit-devices #839: default state when powered on

coyotanark, 2020-03-09:
Hi guys

First thanks a lot for this great FW. 

I have a question, or maybe an enhancement request. I got multiple Sonoffs basic flashed and setup with:

Sonoff Basic R1 | Switch with external toggle | 14: External toggle | {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":14},{"g":14,"t":0}]}]}

I would like those switches to be ON by default when powered on, for example after a power cut. 

I read all https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration

But couldn't find a way to make it. Could you please lighten my path and if not available consider it for future updates?

Thank you

ymyshko, 2020-03-10:
"s": Initial state:
* 0: Default. OFF.
* 1: ON.
* 4: Defined by fixed inputs.
* 5: Last state.
* 6: Opposite to last state.

ymyshko, 2020-03-10:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":1,"s":1,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":14},{"g":14,"t":0}]}]}

coyotanark, 2020-03-10:
Thank you so much you nailed it!

closed 2020-03-10 by coyotanark