RavenSystem/esp-homekit-devices #840: Any way to POST sensor temperature?

frstep, 2020-03-10:
I've just been upgrading my sonoffs to include DS18B20 temperature sensors - is there any way to get the values of these into my influx database? I'm POSTing switch activations which is working very nicely, but would love to record the actual temperatures. (Ideally this would be something easy to do in HomeKit... but alas...!)

Loving HAA - keep up the great work!

peros550, 2020-04-27:
+1 from me, although I understand this is a special feature and may not have such a wide audience

RavenSystem, 2020-04-27:
You can do it using wildcard actions and http requests.

closed 2020-04-27 by RavenSystem

peros550, 2020-04-27:
@RavenSystem many thanks! 
I had missed that, remarkable work!

frstep, 2020-04-27:
Thanks @RavenSystems, II had missed that too. I've seen (and tested) the http request bit, but what are wildcards? I can't see any reference or examples of this. @peros550, have you got this working?

Thanks!

RavenSystem, 2020-04-27:
There are many undocumented features. Wildcard actions will be documented soon.

frstep, 2020-05-02:
I now see the documentation for Wildcards, but... I'm still unclear how this helps me post the temperature data. I see I can set a threshold that triggers an action and can therefore post that threshold has been reached, but not the actual value of the sensor. Is there a way "c" can take a variable?