RavenSystem/esp-homekit-devices #843: Window Covering wrong state after OTA-Update

Kienz, 2020-03-11:
If I run OTA-Update the state of the window is shown as closed. But the state before the update was open.
It also happens if no update is installed because of the latest version is installed or it can't be installed (#825).
In the wiki the "s" parameter (eg. for switches) is not present.

DanielSergienko, 2020-03-12:
Actually there's no way until now to "save" the last state before reboot for window cover/door/etc until now AS FAR AS I KNOW.
I guess you have to keep ur device "On" state all the time. Got the same problem actually. Would love to save last state too.
BUT, also, as RavenSystem said somewhere i dont remember where:
> You MUST enter setup mode with curtains CLOSED.

teymur1988, 2020-03-14:
I also reported that in #807 . Maybe @RavenSystem  can shed some light into it. Another thing that I also reported in #807 is that if you repedeatly keep pressing the icon in homekit to enter setup mode it will enable both of the relays! So make sure you have nothing connected. I've taken the config from the [template ](https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates)and only changed opening and closing time and also added a stop. I mean there should be a way of saving the last state. I read that in Homekid firmware he saves it in EEPROM and then reads it back from there. But I like HAA better, it's much more advanced.

RavenSystem, 2020-03-14:
In Window Covering accessory, last state is always saved 5 seconds after cover stops. But saved states are reseted when you click “Save” button on setup mode. If you want to update manually, close cover first, because default cover state is closed when there is not previous saved state.

closed 2020-03-14 by RavenSystem