RavenSystem/esp-homekit-devices #845: Shelly 1 Garage Door Opener with 1 magnetic reed contact on SW

gerdy001, 2020-03-12:
Hello. Could anyone help me with the json?

I will wire the Shelly on the Garagedoor Motor with 24V DC 
and I will take the fully closed input from a magnetic reed contact (NC) 
(or if it works from the endpoint relais of the Motor)  to the Shelly "SW" input (GPIO5)

lickonline, 2020-03-12:
Hi, I agree with this issue .... also look for a working json for a Hörmann Garagedoor, 
wired to the shelly for control to I and O and power supply with 24V (to L(-) and N(+) 
he wired NC reed contact (to SW and L(-)
    
     (I also tried with tasmota firmware, found also no working plugin/json
     and I had additionally the problem that the GPIO5 input "SW" is
     not working as expected (even not woriking with a virtual relais
      and/or a pullup resistor)

fIctioN600, 2020-03-12:
{"a":[{"t":40,"d":18,"f3":[{"g":5,"t":0}],"f4":[{"g":5}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}}]}

One wire of the reed to "SW" the other to "L"

24V + to "N" - to "L" maybe you need an external 24V DC adapter because if your garagedoor motor goes into standby your shelly loses power too

gerdy001, 2020-03-12:
Thank You very much! I'll test it and give a feedback.

gitigit6, 2020-03-14:
Thanks I'll test it and give a feedback.
(What's the problem at all with the GPIO5 ?)

lickonline, 2020-03-19:
yes, I used GPIO 1 and it works .......... thanks!

(PS: but unfortunately Homekit automations don't work  ... any idea?)

fIctioN600, 2020-03-19:
Do you have an AppleTV, HomePod or iPad configuered as homebase ad connected to power supply?
Automations will not work without a homebase.

gerdy001, 2020-03-20:
With GPIO 1 is this the correct JSON ?
 
{"a":[{"t":40,"d":18,"f3":[{"g":1,"t":0}],"f4":[{"g":1}],"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}}]}

fIctioN600, 2020-03-20:
Yes ist is

gerdy001, 2020-03-21:
Ok. Thank you! It works fine!

closed 2020-03-23 by gerdy001

sercontri, 2022-02-19:
Hello, would I need this plugin for shelly 1 https://shellyspain.com/index.php?id_product=317&rewrite=shelly-addon-temperatura-11-pm&controller=product&id_lang=1 if what I want is to use the SCA connector of a NICE Robus 600 motor to tell me if the door is open or closed? I've thought about putting shelly 1 on the PP connection so that it opens and closes, but I wanted to use the SCA to tell me when it's open. Thank you very much for the help.
[image]