RavenSystem/esp-homekit-devices #846: how to flash with DIY tool Sonoff mini

mitchoklemen, 2020-03-12:
How to flash with DIY tool Sonoff mini over wifi with ESP homekit? howto flash all 3 bin files? haaboot.bin, rboot.bin and blank_config.bin.
Any video to watch?
Thanx, Mitcho

hillaliy, 2020-03-13:
truncate -s 4k rboot.bin
truncate -s 4k blank_config.bin
cat rboot.bin blank_config.bin haaboot.bin > thirdparty.bin

ymyshko, 2020-03-14:
I tried.
DIY tool write "..firmware must be less 506kb..." and don't  flash.
Only soldering? :(

RavenSystem, 2020-03-14:
What is the size of your file?

mitchoklemen, 2020-03-14:
Raven... can U do that BIN file that can be flashed with DIY tool?
That would be great. Thanx for Your work!

RavenSystem, 2020-03-14:
See comment: https://github.com/RavenSystem/esp-homekit-devices/issues/846#issuecomment-598875071

rispie, 2020-03-15:
I tried this earlier today using these instructions: https://blog.bilka.sk/how-to-flash-sonoff-mini-in-mac-os/ and the comment above. All seem to went fine, however I don't see a HAA- AP. So I guess something went wrong. I also read some other issues and comments and there it was mentioned that DIY mode was not supported. So I was wondering if it's supported and if so what did I do wrong?

ymyshko, 2020-03-15:
> What is the size of your file?

420k

ymyshko, 2020-03-15:
I flashed tasmota early. But there is spesial version tasmota to flash sonoff mini thru DIY mode.
tasmota-wifiman.bin 477k
We can't to flash tasmota-minimal 380k

closed 2020-04-14 by stale[bot]

xrust83, 2022-03-15:
> so, it's now possible to flash HAA using this method?
> 
> 1. SONOFF DIY-> TASMOTA
> 2. TASMOTA -> HAA
>    if the answer is yes, @RavenSystem can pin this issue?!

https://www.youtube.com/watch?v=q3sY5w66qeY