RavenSystem/esp-homekit-devices #848: Garage door : 1st press open > 2nd press stop > 3rd press close

Codkit, 2020-03-13:
Hi 
Can anyone teach me the right json for my garage door ?

The physical switch of the garage door operate like this:
When door closed - 1st press open > 2nd press stop > 3rd press close
When door opened - 1st press close > 2nd press stop > 3rd press open

How to translate that to json ? Thanks everyone .

Skankyhank, 2020-03-18:
I use this on my shelly1 for the pulse to open/close the garage door:

{"a":[
{
"t": 1,
"d": 1,
"0":{"r":[{"g":4}]},
"1":{"r":[{"g":4,"v":1,"i":1}]},
"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

It's a switch that returns to off afer 1 second ( "i":1 ) . I do not use the GPIO5 for the physical button, but just connected it in parallel over the shelley relay pins so that in case the shelly dies I can always keep using the physical button.

Codkit, 2020-03-19:
Thanks skankyhank

This only shows a switch , not a garage door t40 ?

Skankyhank, 2020-03-19:
Yes, nothing fancy here. :) If you know how to do that, please let me know.

Codkit, 2020-03-19:
To be honest i really have no clue , i tried the sample json from the wiki with emergency stop , the door switch in home app doesn’t work after clicking the emergency stop button.

acidmad, 2020-03-31:
> I use this on my shelly1 for the pulse to open/close the garage door:
>
> {"a":[
> {
> "t": 1,
> "d": 1,
> "0":{"r":[{"g":4}]},
> "1":{"r":[{"g":4,"v":1,"i":1}]},
> "b":[{"g":5,"p":0},{"g":5,"t":0}]}]}
>
> It's a [...]

Can reverse signal GPIO5, 
In this embodiment, when the power is turned off and on, the door opens and closes.

closed 2020-04-30 by stale[bot]