RavenSystem/esp-homekit-devices #849: Sonoff S55 Support

buarle, 2020-03-14:
Hi and thanks for this awesome work.

Could you please let me know, if it's possible to flash a Sonoff s55 with your software?
Would be really nice to use this outdoor (IP55) outlet via Homekit.

Thanks a lot and BR!
Buarle

buarle, 2020-03-14:
It's build with an ESP8285.
Some information to this device:
https://fccid.io/2APN5-S55/Internal-Photos/Int-Photos-4013503

stefantitu, 2020-03-15:
I flashed a magichome led strip controller that has esp8285, and it works ok. So it should also work on sonoff s55

buarle, 2020-03-26:
I ordered one and will tell you the result!

closed 2020-04-25 by stale[bot]

Bogdis, 2021-05-02:
> I ordered one and will tell you the result!
Hi, did you manage to flash the outlet? If so can you tell me how you did it?
Thanks