RavenSystem/esp-homekit-devices #851: Add another switch to my Homekit, ”Device already added”

Skankyhank, 2020-03-19:
Using the same JSON as in the other switch that i’m using.

The thing is that when I try to set it up, the already installed switch stops working. And as soon as I cut the power to the one i try to set up, the already installed one starts to work again.

{"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

dfrysinger, 2020-04-05:
I am seeing the same problem trying to add a second MagicHome RGBW Controller. Do we need different setup codes for each? Is that configurable?

Skankyhank, 2020-04-06:
For me it’s working right now. I belive it was due to the OTA update issue.

Use the latest haboot.bin and you should be ok.

closed 2020-04-06 by Skankyhank