RavenSystem/esp-homekit-devices #853: Inching timer should stop on second press

mmiglier, 2020-03-21:
Hi, thanks again for your great work. I configured my roller shutter with a shelly 2.5 and the following configuration:

```
{
    "c": {
        "l": 0,
        "n": "tapsala"
    },
    "a": [
        {
            "0": {
                "r": [
                    {
                        "g": 15
                    },
                    {
                        "g": 4,
                        "v": 1
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 15,
                        "v": 1
                    },
                    {
                        "g": 4
                    }
                ]
            },
            "2": {
                "r": [
                    {
                        "g": 15
                    },
                    {
                        "g": 4
                    }
                ]
            },
            "3": {
                "r": [
                    {
                        "g": 15
                    },
                    {
                        "g": 4,
                        "i": 1
                    }
                ]
            },
            "4": {
                "r": [
                    {
                        "g": 15,
                        "i": 1
                    },
                    {
                        "g": 4
                    }
                ]
            },
            "t": 45,
            "o": 34,
            "c": 31,
            "f": 30,
            "f0": [
                {
                    "g": 13,
                    "p": 0
                }
            ],
            "f1": [
                {
                    "g": 5,
                    "p": 0
                }
            ],
            "f2": [
                {
                    "g": 5,
                    "t": 0
                },
                {
                    "g": 13,
                    "t": 0
                }
            ]
        }
    ]
} 
```

The problem I have is that if I double press the accessory on the homekit app while closing or opening, the motor will start pulling both up and down. This is not a real use case, but something that could happen by mistake and the risk is to probably ruin the motor.

The reason why that happen is that the the second press does not disable the previously started timer for the inch, I guess. I could decrease the inching time to, say, 100ms, since a double tap time should be longer. But this could damage the motor as well.

My question is: shouldn’t the second press override the first press and any inch timer started by it?

mmiglier, 2020-03-22:
I'll try to explain better.

Actual behavior (as far as I understood):
```
time 0.0s - "3": {"r": [{ "g": 15 },{"g": 4,"i": 1 }]} -> g15 is set to 0 and g4 is set to 0
time 0.5s - "4": {"r": [{ "g": 15, "i": 1  },{"g": 4 }]} -> g15 is set to 0 and g4 is set to 0
time 1.0s - g15 is still 0 and g4 is set to 1 as a consequence of action 3
time 1.5s - g15 is set to 1 as a consequence of action 4 and g4 is still 1
```
as a result both g15 and g4 are set to 1.

Expected behavior:
```
time 0.0s - "3": {"r": [{ "g": 15 },{"g": 4,"i": 1 }]} -> g15 is set to 0 and g4 is set to 0
time 0.5s - "4": {"r": [{ "g": 15, "i": 1  },{"g": 4 }]} -> g15 is set to 0,  g4 is set to 0 and inch timer is stopped
time 1.0s - g15 is still 0 and g4 is still 0
time 1.5s - g15 is set to 1 as a consequence of action 4 and g4 is still 0
```

mmiglier, 2020-04-19:
Hi @RavenSystem did I explain the problem clearly to you?

closed 2020-05-19 by stale[bot]

gregory90, 2020-09-03:
@RavenSystem Why is this issue marked as stale? It could possibly fry roller shutter motor. What can I do to move this issue forward? It's been a while since I've been writing C, but if you tell me what part of code is responsible for this I could try to solve this issue. Thanks in advance