RavenSystem/esp-homekit-devices #856: Special Template for 2 Motion Sensors with a Shelly 2.5

Hanimaniac, 2020-03-22:
Hello Community,

I use the HAA System and it‘s awsome. I also made my own Templates but for my next Project i‘m not fit enough. 
Is it possible to use a Shelly 2.5 as „2“ Motion Sensors? So that each Sensor triggers at one Switch input?
[image]

Like in my sketch I‘d like to use 2 230V Sensors which trigger the Shelly and in Homekit it should attach as 2 Motionsensors. 
I hope you can help me. 
Greetings from Germany. 
Hanimaniac

Kienz, 2020-03-23:
```json
{"c":{"l":0,"b":[{"g":2,"t":5}]},"a":[{"t":12,"i":10,"f1":[{"g":5,"p":0,"i":1,"t":0}],"s":0},{"t":12,"i":10,"f1":[{"g":13,"p":0,"i":1,"t":0}],"s":0},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}
```
Two motions sensors and two virtual switches to enter setup mode and start ota-update.
I've tested multiple PIR sensors - for me only the one mentioned in #741 are working.

Hanimaniac, 2020-03-23:
Thanks for your fast Answer. I‘ll test it soon.

Kienz, 2020-03-27:
@Hanimaniac Does it work?

Hanimaniac, 2020-03-27:
Sorry, I forgot to report 🙈
It works perfectly with my two pir Sensor.
I use these:
 https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F123703914364
Thank you

closed 2020-03-27 by Hanimaniac

Hanimaniac, 2020-04-20:
@Kienz 
Can I contact you pm?

Kienz, 2020-04-20:
> @Kienz
> Can I contact you pm?

Yes.
Twitter @kienzle_s

Hanimaniac, 2020-04-20:
Ich hab kein Twitter 🙈

Kienz, 2020-04-20:
@Hanimaniac How can I help you?

Hanimaniac, 2020-04-21:
It‘s off topic. 
I use ca. 20 haa/ ravencore device in my wifi. 
I have a Fritzbox 6591 wit a fritz repeater. 
Fron time to time i have no response from my devices. completly random. no specific typ. its shelly 1 Shelly 2.5 and Magic Home LED Controller. 
So my automations sometimes dont work because a device is not responding. 
I‘d like to ask you about your wifi settings and if you may have simillar problems.

Kienz, 2020-04-21:
I'm using UniFi APs and have this issues if the device connects to a poor AP. Since I select a fix AP in HAA wifi config I don't have issues. Currently I'm using 18 HAA devices (Shelly 1, Shelly 2.5 and Blitzwolf BW-LT21).

Hanimaniac, 2020-04-22:
ok. thanks for your answer.