RavenSystem/esp-homekit-devices #857: Sonoff RS with external toggle

danielssantos, 2020-03-22:
I'm using Sonoff R2, with external toggle in GND and IO2. On RavenCore works perfectly. On HAA, the external toggle don't work. Using this template, what's wrong?

{
"c":{"o":1,"l":13,"i":1},
"a":[{
"t":1,
"0":{"r":[{"g":12,"v":0}]},
"1":{"r":[{"g":12,"v":1}]},
"b":[{"g":0,"p":1,"t":1},
{"g":14,"p":1,"t":1},
{"g":14,"t":0}]
}]
}

xtremedays, 2020-03-24:
I use this with toggle switches

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r": [{"g":12,"v":1}]},"b":[{"g":0},{"g":2},{"g": 2,"t":0}]}]}

danielssantos, 2020-03-24:
Thank you very much! It's works perfectly with my sonoffs R2 with GPIO2. For old versions with GPIO14, tried change your code without success. But works with this template: 

{"c":{"o":1,"l":13,"i":1},"a":[{"t":1,"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0,"p":1,"t":1},{"g":14,"p":1,"t":1},{"g":14,"t":0}]}]}

closed 2020-04-24 by stale[bot]