RavenSystem/esp-homekit-devices #860: iFan03 433mhz remote not working

Codkit, 2020-03-27:
Hi all

Anyone possible to teach me how to make it work ? Thank you.

pilot1981, 2020-03-27:
also me interested...you are using it with sonoff basic RF R3?

Codkit, 2020-03-27:
hi
I mean the original RF remote with iFan03

hillaliy, 2020-03-30:
To pair the remote control, hold down any button on the remote control and apply power to the iFan03. Once the device boots up you should hear a series of clicks as the internal relays operate. Once completed you can test the remote and it should all be operational.

Codkit, 2020-03-30:
Hi
It doesn’t work after flashing the firmware

hillaliy, 2020-03-30:
it's work after tasmota flash

fiska1, 2020-04-01:
Hello.  I have the same problem.  The remote does not work.  Has anyone made it work after firmware?

Jeroen207, 2020-04-12:
So, the remote seems to work with some people here?

400HPMustang, 2020-05-04:
I'm also interested in how to get the remote working. I flashed an iFan03 and cannot get the remote to work.

RavenSystem, 2020-05-19:
iFan03 remote will no be supported soon.

closed 2020-05-19 by RavenSystem