RavenSystem/esp-homekit-devices #862: Shelly Plug S Json

OnKram, 2020-03-29:
Hi all,
I have trouble with Json for Shelly Plug S. 
If I connect device to Home App so it works fine, but manual switch not working.
I used GPIO pins from this page: https://templates.blakadder.com/shelly_plug_S.html and I also try others pins but still not working it.

Any idea please?
{"c":{"l":0,"b":[{"g":13,"t":5}]},"a":[{"0":{"r":[{"g":15},{"g":2,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":2,"v":0}]},"t":2,"b":[{"g":13,"t":1,"i":0,"p":0}],"s":0}]}

RavenSystem, 2020-03-29:
"i":0,"p":0 must be set at the first declaration of GPIO to be used. Try moving them.

OnKram, 2020-03-29:
Ok, I tried this setup, but still not working.

{"c":{"l":0,"b":[{"g":13,"t":5,"i":0,"p":0}]},"a":[{"0":{"r":[{"g":15},{"g":2,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":2,"v":0}]},"t":2,"b":[{"g":13,"t":1}],"s":0}]}

closed 2020-04-28 by stale[bot]