RavenSystem/esp-homekit-devices #863: Provoke no response

asbesto01, 2020-03-30:
Hello everybody, I am trying to deeply test haa ravencore reliability on sonoff flashed devices.

I tried this test:

Quickly turn on ,then off, then on again (0.5 seconds delay among each turn on and off) a sonoff basic with haa flashed (it controls a light).

Well, after 6 or 7 quick on and off the device goes in NO RESPONSE, then it comes back after some minutes.

Is this behavior normal? I have never had similar issues on zigbee HomeKit switches for example.

Is it due to poor router performances? Poor sonoff performances? Haa architecture?

Thank you!!!!

Bolshoylu, 2020-03-30:
maybe it goes into setup mode?

Entering Setup Mode

To enter setup mode, you must quickly toggle any action switch/button 8 times (physical or from Home App). You can secure this method by declaring a time after device boot to perform it.

Additionally, you can declare a physical method to enter setup mode.

Emergency Setup Mode: If a device has its power cut or freezes within 3 seconds of boot, next time it boots it will go directly to setup mode.

RavenSystem, 2020-03-30:
> Well, after 6 or 7 quick on and off the device goes in NO RESPONSE, then it comes back to life after some minutes.

@asbesto01 Totally right. Read this: https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#entering-setup-mode

> The bad thing is that after the triggered device goes in NO RESPONSE, other sonoff haa flashed devices go in no response too, even if I have never activated them!!

That is normal, because HomeKit communications are concatenated.

> I have never had similar behavior on zigbee HomeKit switches for example.

Zigbee is not HomeKit. Normally, Zigbee devices use a HomeKit bridge.

peros550, 2020-03-31:
> That is normal, because HomeKit communications are concatenated.
> 
So if one device goes to no response for whatever reason, it could take with it all the others? This is something I have experienced also. 

But why this is happening only with HAA or custom firmware devices? I have for example Philips Hue bulbs which are never shown as no response .

asbesto01, 2020-03-31:
Ciao Peros, as Jose said, zigbee devices (or generally speaking devices with an hub) are not affected

Philips hue are zigbee

Jose muchas gracias por todo, espero Que espana va vencer esto virus

Salutos desde Italia

closed 2020-04-30 by stale[bot]