RavenSystem/esp-homekit-devices #865: Too much accuracy in Nome Assistant

xrust83, 2020-03-31:
I use sonoff basiс + Si7021 but the firmware gives too much accuracy for the sensor 22.2000007629395 °C temp and 53.5000006543235 % humidity. Can you round to 0.1?

xrust83, 2020-03-31:
All is well in HomeKit. But I connected the device to the phone all through HomeKit Home Assistant well. Tell me, maybe you know how to fix the display in the Home Assistant. How to round. In Home Assistant it changes sometimes very precise temperature and humidity up to ten times the other way around

RavenSystem, 2020-03-31:
Sorry, I personally don’t use Home Assistant for security reasons. Maybe other people can help you, but HAA uses float type to store raw temperature data, and way to work is right with official HomeKit specifications.

xrust83, 2020-03-31:
I have one more question for you. I use Sonoff Dual as a 2-bulb chandelier switch.
I wanted to update them. Tell me how to enable the operation mode of external buttons with fixation in HAA firmware. External Toggle was selected in the equalizer (2)

closed 2020-04-09 by xrust83