RavenSystem/esp-homekit-devices #866: Problem withe connect to HAA-xxxxxx

mientki, 2020-04-01:
Is there a known problem with the current version of HAA? I have a problem connecting to the HAA-xxxxxx network after flashing. I have flashed many different devices many times and in all cases there is the same problem. Cannot connect to the created HAA-xxxxxx network.

RavenSystem, 2020-04-01:
Logs, please.

Kienz, 2020-04-01:
Since macOS 10.15.4 the captive portal is opening but the HAA page is not loading. In macOS 10.13.6 it works. Has anyone the same problem?

mientki, 2020-04-01:
I have the same problem after upgrading mac os to version 10.15.4. The window opens but has no content. On Mac os 10.13.6 it's OK.

mientki, 2020-04-01:
```
scandone
scandone
scandone
add 1
aid 1
station: dc:a9:04:75:62:92 join, AID = 1
Got 64 incomming data
! Unknown: GET /hotspot-detect.html
Got 64 incomming data
Got 3 incomming data
Unknown
Redirecting to http://192.168.4.1/settings
Client disconnected
Got 64 incomming data
Got 50 incomming data
Client disconnected
Got 64 incomming data
! Unknown: GET /hotspot-detect.html
Got 64 incomming data
Got 64 incomming data
Got 64 incomming data
Got 64 incomming data
Got 27 incomming data
Unknown
Redirecting to http://192.168.4.1/settings
Client disconnected
scandone
Got 64 incomming data
Got 64 incomming data
Got 64 incomming data
Got 64 incomming data
Got 64 incomming data
Got 10 incomming data
Client disconnected
scandone
scandone
```

arturodiazad1, 2020-04-01:
I am having the same problems that you mention, also some devices loop when they are downloading packages, I use version 10.15.4 and with two different mac I have the same problem

wwns, 2020-04-02:
Everything is ok wait it will show you the configuration page

RavenSystem, 2020-04-02:
Fixed with last versions.

closed 2020-04-02 by RavenSystem