RavenSystem/esp-homekit-devices #867: No Response on esp8266

hecguillem, 2020-04-01:
First of all congratulations for the firmware.
I have configured the RavenSystem in esp8266, wemos .. and I have noticed that they frequently go to "not responding", then return to normal after 5-10 min, devices are close to the router and good signal. Could you find out the reason for the falls? Any log or something? thanks (from Spain)

RavenSystem, 2020-04-02:
HAA has not any known "no response" issue since all measures included in v1.7.0 to mitigate mDNS bad effects. Of course, logs are needed to debug. What JSON are you using? Have you got any other esp8266/esp32 device using original esp-homekit library?

hecguillem, 2020-04-02:
628/5000
I can contribute Json but not log because the USB of my PC works perfect!
JSON: {"a": [{"t": 20, "w": 1, "d": 3600, "0": {"r": [{"g": 4}]}, "1" : {"r": [{"g": 4, "v": 1}]}, "b": [{"g": 5, "p": 0}, {"g": 5, " t ": 0}]}, {" i ": 0.5," 1 ": {" s ": [{" a ": 1}]}}]}
My case is as follows, when I connect my esp8266 to 5V (different chargers and cables) homekit shows it to me as "not responding", but when I connect it to the USB of my Macbook Air it is stable and has a quick response. If I connect to a windows computer or any power source it fails. What could this be? I have more esp8266 (shelly) and no problem but they are at 220V.

Thanks for your help

hecguillem, 2020-04-02:
I think that I have solution.

I have tested other esp-librarys such esp32cam of maxim, Homekidd projects, and your firmware.

Problem is esp8266 or other esp are unstable at 3,3V.. is my experience. 

Thanks for all!

closed 2020-04-02 by hecguillem

RavenSystem, 2020-04-02:
Esp8266 works with 3.3v, not more, and max allowed voltage is 3.6v but it is not recommended.