RavenSystem/esp-homekit-devices #868: OTA

Kristian8606, 2020-04-02:
Hello my english is bad and i can't figure out exactly how to compile the three files into one to flash the firmware using tuya convert.  can i flash haa main.bin file without LCM with tuya convert

wwns, 2020-04-02:
hexeditor

tonysprenk, 2020-04-02:
https://github.com/ct-Open-Source/tuya-convert/wiki/Flash-a-multipart-binary

Kristian8606, 2020-04-02:
Thanks the problem is that when I place a bin file in the directory and start the flashing procedure, the bin file cannot be selected just doesn't show

tonysprenk, 2020-04-02:
Are you sure you put it in the right directory?

Kristian8606, 2020-04-02:
Directory /files

tonysprenk, 2020-04-02:
Hmm it should work than. Maybe open an issue on the tuya-convert repo.

Kristian8606, 2020-04-03:
i suspect that bin is wrong someone can give me bin

Kristian8606, 2020-04-03:
https://github.com/Kristian8606/LED_Strip/releases/download/4.0.2/thirdparty.bin

the problem was in the bin file. I was able to build working with the last version of HAA

closed 2020-05-03 by stale[bot]