RavenSystem/esp-homekit-devices #869: iFan03-It doesn't work correctly

fiska1, 2020-04-02:
Hello. After firmware and JSON installation from the database, the fan does not work correctly. At 33% - the maximum speed, at 100 %-the minimum. How do I fix it ?
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"e":3,"0":{"r":[{"g":15},{"g":12},{"g":14}]},"y0":[{"v":1,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12},{"g":14,"i":0.6}]}},{"v":2,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12,"i":0.6},{"g":14,"i":0.6}]}},{"v":3,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":14}]}}]},{"1":{"r":[{"g":9}]},"0":{"r":[{"g":9,"v":1}]},"b":[{"g":0}]}]}

RavenSystem, 2020-04-02:
Try this and comment results, please:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":65,"e":3,"0":{"r":[{"g":15},{"g":12},{"g":14}]},"y0":[{"v":3,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12},{"g":14,"i":0.6}]}},{"v":2,"0":{"r":[{"g":15,"v":1,"i":0.5},{"g":12,"i":0.6},{"g":14,"i":0.6}]}},{"v":1,"0":{"r":[{"g":15,"v":1},{"g":4},{"g":14}]}}]},{"1":{"r":[{"g":9}]},"0":{"r":[{"g":9,"v":1}]},"b":[{"g":0}]}]}

fiska1, 2020-04-02:
Thanks, now it works correctly. I don't use it for ceiling fans, but for exhaust fans.Can't the code fix the speed? 100% rotates very strongly (non-standard).. Without the iFan 03 device the maximum rpm is 67%

Partizan7676, 2020-04-07:
@fiska1 does RF-remote works too? when you use it is status changed in home app ???

fiska1, 2020-04-07:
No.  The remote does not work.

Gwyki, 2020-04-08:
Apologies for jumping into this issue, but I don't think a new one is warranted for my questions...

@RavenSystem Can you provide a quick bit of documentation for these settings here? The small hint in the release notes doesn't cover the accessory specific action variables like the "i" here?

Should home app be displaying this as a continuous bar, or the segmented switch? I am getting a continuous slider...

Jeroen207, 2020-04-10:
Is it possible to add RF remote capability with json?

closed 2020-05-10 by stale[bot]

i3laze, 2020-05-11:
> Apologies for jumping into this issue, but I don't think a new one is warranted for my questions...
>
> @RavenSystem Can you provide a quick bit of documentation for these settings here? The small hint [...]

@Gwyki , Wiki is at your disposal:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Fan

bj-tracer, 2020-06-15:
Hello!
Can it control with external buttons?

brymonster, 2020-07-03:
@fiska1 did u get the ifan03 working with an external wired switch?