RavenSystem/esp-homekit-devices #870: Support request

Kristian8606, 2020-04-03:
Is it possible to add support for this bulb

[image]
[image]
[image]
[image]

tonysprenk, 2020-04-03:
The chip is covered by the white glue so it’s impossible to see if it’s ESP based. Easiest way to find out is to google the MAC adres of the device. It will tell you who the manufacturer of the chip is. Post the results when you have them.

tonysprenk, 2020-04-03:
Looking at the pictures it looks like this is not an ESP chip but rather a newer chip that tuya has been putting into their devices. It’s based around an ARM micro controller. But to know for sure google the Mac adres or remove the glue and share a picture.

Kristian8606, 2020-04-03:
The chip is esp8266 I managed to flash it HAA

tonysprenk, 2020-04-03:
Oh ok that’s good! Try this json: 

```

{"c":{"b":[{"g":0}]},"a":[{"t":30,"r":14,"g":12,"v":13,"w":4,"fr":4,"fg":4,"fv":4}]}

```

It’s from the device database innige wiki.

Kristian8606, 2020-04-03:
Did not work. I don't understand how this json works but as far as i understand esp manages this driver through sda clk pins

Kristian8606, 2020-04-03:
uses pin 4 and 14 of esp which are connected to the led driver

tonysprenk, 2020-04-03:
So it sends serial commands to a secondary micro controller? This feature is not yet enabled in HAA but @RavenSystem is working on it.

Kristian8606, 2020-04-04:
https://git.koehlerweb.org/frodovdr/Sonoff-Tasmota/src/commit/6997626077c18b11f2c2e672b9ba8e88f37c9046/sonoff/xdrv_26_sm2135.ino

closed 2020-05-04 by stale[bot]