RavenSystem/esp-homekit-devices #871: JSON readability suggestion

i3laze, 2020-04-04:
As @RavenSystem told million times there's little space in the Flash to store all the "Relay" "Accessory" and other Words of the JSON config, that is why shortened "l","e","t","t","e","r","s" were introduced.

Every time once a year when I try to reconfig HAA, I'm down for an hour into Wiki.
It's always hard to decipher JSON, and https://glumb.github.io/haa-configurator/ barely helps.

Suggestion:
Please, introduce alternate JSON view (some checkbox on HAA Setup webpage) with words exploded to "Accessory", "Type", "Relay" and then strip them on the fly back to "a","t","r" when saving to flash.

Example - empty accessory:
`{"c":{},"a":[]}`
Same example with explode checkbox On:
`{"Config":{},"Accessory":[]}`

Cross posted here:
https://github.com/glumb/haa-configurator/issues/8

i3laze, 2020-04-04:
Just an illustration of Config Readability:
[image]
RavenCore << Tasmota << HAA

freddair, 2020-05-09:
+1 for this!

RavenSystem, 2020-05-19:
JSON is as it because memory and storage are very important. You can use a simple find and replace to create your own tool.

closed 2020-05-19 by RavenSystem

beren12, 2020-07-02:
It would be nice to have the ability to switch back and forth though, feed the tool one and it spits out the other :)