RavenSystem/esp-homekit-devices #876: Shelly 2.5 Step by Step Relay

milonet, 2020-04-06:
Hi, it's possible to make a step by step relay with haa latest firmware on shelly 2.5? Do you have a json?

best regards

kx3kx3kx3, 2020-04-06:
try this:
{"c":{"l":0,"i":1},"a":[{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0,"i":1}]},{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0,"i":1}]}]}

milonet, 2020-04-06:
Hi, thank you but it's not working. Work normally as two
separate switch/outleet.
I need
- First press on input1 -> Outlet 1 On
- Second press -> Outlet 1 OFF, Outlet 2 ON
- Third press -> Outlet 1 On, Outlet 2 On
- Fourth press -> Outlet 1 OFF, Outlet 2 OFF

Like as step by step Relay

Thansk

kx3kx3kx3, 2020-04-06:
> Hi, thank you but it's not working. Work normally as two separate switch/outleet. I need - First press on input1 -> Outlet 1 On - Second press -> Outlet 1 OFF, Outlet 2 ON - Third press -> Outlet 1 [...]

On home app or on a physical button?

milonet, 2020-04-06:
On physical button

kx3kx3kx3, 2020-04-09:
try this one  : 
{"c":{"l":0,"b":[{"g":2,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0,"i":1,"t":0}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0,"i":1,"t":0}]}]}
 
and let me know

milonet, 2020-04-09:
thank you but it's still not working.. the switch/outles works independently

best regards

kx3kx3kx3, 2020-04-09:
if you want to try another json for shelly 2.5
{"c":{"l":0,"b":[{"g":2,"p":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}],"s":0},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0} ],"s":0}]}

milonet, 2020-04-12:
it's not working. with this configuration the outlets not keeping the state. when release the physical button the outlets goes off

closed 2020-05-12 by stale[bot]