RavenSystem/esp-homekit-devices #877: ESP32-Cam

egorist, 2020-04-06:
Are there any plans to bring HAA into ESP32-Cam module?
It will be good to make working in HomeKit not only camera, but led, sensors and another possible modules. 
I can test, if you’ll try to make firmware. 
There is some information:
https://github.com/maximkulkin/esp32-homekit-camera
[image]

JayJayXPro, 2020-04-30:
The module could someday work as an element of a door bell.  I'm looking forward!

tonysprenk, 2020-04-30:
The problem with this is that the esp32 doesn’t have enough computing power to convert the camera stream to a format that HomeKit is happy with resulting in a very low frame rate (like 6 or 7 fps). Look on YouTube for some examples, it doesn’t look very usable. A better solution is [HKCam](https://github.com/brutella/hkcam).

This could be combined with a simple HAA button and some scenes/automations to create the effect of a doorbell, as the doorbell accessory type is not supported by HAA.

egorist, 2020-04-30:
> The problem with this is that the esp32 doesn’t have enough computing power to convert the camera stream to a format that HomeKit is happy with resulting in a very low frame rate (like 6 or 7 fps). [...]

I use esp32cam. It works via rtsp and homebridge perfectly for my needs. But this module can be more effective with HAA and ability to have video stream and to program gpios (light, buttons,sensors, locks and so on)

tonysprenk, 2020-04-30:
Yes it works well with homebridge because the computer your running your homebridge server on is doing the encoding of the rtsp stream. When you take away that computer like with HAA there is not enough computing power to convert the rtsp stream to a usable format for HomeKit.

egorist, 2020-04-30:
But it’s possible to make HAA for gpios + video via homebridge, isn’t it?😄

JayJayXPro, 2020-04-30:
> But it’s possible to make HAA for gpios + video via homebridge, isn’t it?😄 

Good question.  Even if the vision was to have 7fps.

egorist, 2020-04-30:
It’s time to start HAA on esp32, even without video😄

JayJayXPro, 2020-04-30:
I would like to make a video doorbell compatible with HomeKit based on this element.  That would be a great deal

egorist, 2020-04-30:
Yes, because often only picture is needed

JayJayXPro, 2020-04-30:
> Yes, because often only picture is needed 

Exactly.  It's enough for the camera to take a photo when calling.

JayJayXPro, 2020-05-01:
So please tell me what solution would be best and compatible with HomeKit. I would like to build a video doorbell cooperating with the lock mechanism (Sonoff SV 12V). I want to have a notification about the call, a photo or (if it possible) live view. The most important thing is possiblility to communicate with a person on the other side. It would be blow my bind.

peros550, 2020-05-01:
+1 for the video doorbell!

egorist, 2020-05-01:
> So please tell me what solution would be best and compatible with HomeKit. I would like to build a video doorbell cooperating with the lock mechanism (Sonoff SV 12V). I want to have a notification [...]

As I see, now you need 2 devices (if you want to use sonoff sv). 
1 - camera via homebridge (I don’t know about ability to talk with person, but you can hear, exactly)
2 - sonoff sv to open the door and free gpio for button.

To use 1 device, you need homebridge server running on something like raspberry (for video surveillance I use windows 10 pc and some outdoor Reolink cameras with speakers. If I need to communicate, I use official reolink app. HomeKit - for “see and open” case). 
2 device must be configured in HAA setup page like 2 accessories: lock and button (but I use xiaomi motion sensor because of HomeKit notifications with picture. But I think, in theory, in HAA you may declare button as motion sensor).

maxplex, 2020-05-15:
Could someone clarify if flashing 'fullhaaboot.bin' to an ESP32 will work? I'm a noob... Be gentle ;)

Kienz, 2020-05-15:
> Could someone clarify if flashing 'fullhaaboot.bin' to an ESP32 will work? I'm a noob... Be gentle ;)

No. See https://github.com/RavenSystem/esp-homekit-devices/issues/447#issuecomment-508794927

RavenSystem, 2020-05-19:
HAA does not work on an esp32.

closed 2020-05-19 by RavenSystem