RavenSystem/esp-homekit-devices #879: Ravencore to HAA full copy of external button functions

xrust83, 2020-04-09:
help configure sonoff dual with HAA firmware to work like ravencore
[image]

RavenSystem, 2020-04-14:
`{"c":{"l":13,"b":[{"g":10,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":0,"t":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9},{"g":9,"t":0}]}]}`

xrust83, 2020-04-14:
It works. Thanks = -D. Will I be able to update all the switches by air or will I have to remove them from the ceiling? I already compared both codes in the editor https://glumb.github.io/haa-configurator/ and figured out how to install it

closed 2020-05-14 by stale[bot]