RavenSystem/esp-homekit-devices #880: Scandone Reconect Bootloop

Zoiko192, 2020-04-11:
I have a loop after config the JSON and Wi-Fi password without OTA. I went up several times before but now I get this error:

```
...
lwip_read 3
Update settings, body = conf=%7B%22c%22%3A%7B%22l%22%3A16%2C%22n%22%3A%22sonoff%22%2C%22o%22%3A1%2C%22b%22%3A%5B%7B%22g%22%3A0%2C%22t%22%3A5%7D%5D%7D%2C%22a%22%3A%5B%7B%220%22%3A%7B%22r%22%3A%5B%7B%22g%22%3A16%7D%5D%7D%2C%221%22%3A%7B%22r%22%3A%5B%7B%22g%22%3A16%2C%22v%22%3A1%7D%5D%7D%2C%22b%22%3A%5B%7B%22g%22%3A5%7D%2C%7B%22g%22%3A14%7D%2C%7B%22g%22%3A14%2C%22t%22%3A5%7D%5D%7D%5D%7D&wifimode=0&bssid=14edbb2f3d6d&ssid=MissyCX&password=Mipersona%23%2482
Rebooting...
del if0
station: 4c:8d:79:d7:93:dc leave, AID = 1
rm match
bcn 0
del if1
usl
sul 0 0
ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b
rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data
Booting rom 0.
WAdå|˚cÏÛo˚pÇÑooülnoÏòn‡pg‰s#ìn| coådc`{lpp_task_hdl : 3ffefa38, prio:14, stack:512
pm_task_hdl : 3ffef4a0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4608, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr  2 2020 16:00:33
phy ver: 273, pp ver: 8.3
Sysparam OK
WiFi Init
WiFi Mode: Normal
mode : sta(ec:fa:bc:63:2e:10)
add if0
scandone
reconnect
scandone
reconnect
```

DanielSergienko, 2020-04-12:
Bad json. Go to local ip of your NodeMCU

Zoiko192, 2020-04-13:
The esp8266 does not connect to my router.  Restart my router and it was fixed.

closed 2020-04-13 by Zoiko192