RavenSystem/esp-homekit-devices #882: shelly 2.5 windowcovering external switches JSON wrong?

Leonoerlemans, 2020-04-11:
Hi many many thanks for this great work.

I have an issue with the shelly 2.5 in window covering mode with EXTERNAL switches
I can flash it works fine I can get it to work with homekit just fine in double relay mode and window covering mode with pushbuttons. When I use the json from te he example I can't homekit to recognise it. I'm pretty sure that something in the json is wrong for when I use the json for relays and windowcovering with push buttons it works just fine.

when I compare them, this is what I notice:

in this section: f0":[{"g":5,"p":0}],"f1":[{"g":13,"p":0}],"f2":[{"g":5,"t":0},{"g":13,"t":0}] where the function of the buttons are defined (The ONLY difference) the "t" is used in f2 (instead of the "p" as in f0 and f1)

I tried to change that, but there was no luck...

PLEASE help
Kind regards, Leon

the complete json's:

Shelly2.5 Window Covering driver with external switches, status LED and built-in setup button (Opening time: 18 seconds. Closing time: 14 seconds. Non-lineal correction: 70%):

{"c":{"l":0,"b":[{"g":2,"t":5}]},"a":[{"t":45,"o":18,"c":14,"f":70,"0":{"r":[{"g":15},{"g":4,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":4}]},"2":{"r":[{"g":15},{"g":4}]},"3":{"r":[{"g":15},{"g":4,"i":0.2}]},"4":{"r":[{"g":15,"i":0.2},{"g":4}]},"f0":[{"g":5,"p":0}],"f1":[{"g":13,"p":0}],"f2":[{"g":5,"t":0},{"g":13,"t":0}]}]}

Shelly2.5 Window Covering driver with external push buttons, status LED and built-in setup button (Opening time: 18 seconds. Closing time: 14 seconds. Non-lineal correction: 70%):

{"c":{"l":0,"b":[{"g":2,"t":5}]},"a":[{"t":45,"o":18,"c":14,"f":70,"0":{"r":[{"g":15},{"g":4,"v":1}]},"1":{"r":[{"g":15,"v":1},{"g":4}]},"2":{"r":[{"g":15},{"g":4}]},"3":{"r":[{"g":15},{"g":4,"i":0.2}]},"4":{"r":[{"g":15,"i":0.2},{"g":4}]},"f3":[{"g":5,"p":0}],"f4":[{"g":13,"p":0}]}]}

Leonoerlemans, 2020-04-12:
In addition: I have looked into all items including:

https://github.com/RavenSystem/esp-homekit-devices/issues/761

seemed that th eproblem occured earlier (firmware 1.2.1) but was fixed in a later firmware 1.4. Right now I'm running firmware 1.9.2. Did the problem reappear?

Kind regards

Leon

RavenSystem, 2020-04-12:
Update to v1.9.3 because there was a pairing issue.

Leonoerlemans, 2020-04-12:
thank you. I'll try right away!

Leonoerlemans, 2020-04-12:
Hi,

The new firmware (v 1.10) doesn't do the trick

log:
scandone
rm match
reconnect
scandone
add 0
aid 2
cnt 

connected with UbiquitiLJIJ, channel 6
dhcp client start...
scandone
ip:192.168.2.55,mask:255.255.255.0,gw:192.168.2.254
scandone
scandone

It does connect, but when I try to pair it with homekit Home app says this accessory was already added. I can connect it when I use the json for pushbuttons, therefor I tthink it might be in the json itself.

Has anyone got it to run with external switches in windowcovering mode?

Please help

Kienz, 2020-04-16:
@Leonoerlemans I'm using this json for many windows covers with external switches and all are working fine and can be added to HomeKit without problems.

```json
{"c":{"l":0,"b":[{"g":2,"p":0,"t":5}]},"a":[{"t":45,"o":28,"c":26,"f":70,"0":{"r":[{"g":15,"v":1},{"g":4}]},"1":{"r":[{"g":15},{"g":4,"v":1}]},"2":{"r":[{"g":15},{"g":4}]},"3":{"r":[{"g":15,"i":1},{"g":4}]},"4":{"r":[{"g":15},{"g":4,"i":1}]},"f0":[{"g":5,"p":0}],"f1":[{"g":13,"p":0}],"f2":[{"g":5,"t":0},{"g":13,"t":0}]}]}
```
This json is from the [Wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering#window-covering-example)

@RavenSystem I don't know what the ```"p":0``` parm is for. Can you explain.

Leonoerlemans, 2020-04-16:
@Kienz Thank You!

This json is from the explained sample on wiki. 
I used: https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates
also from wiki, but in the sample section, with more samples. That one is different and does not work.
@RavenSystem This was the isue

closed 2020-05-16 by stale[bot]