RavenSystem/esp-homekit-devices #883: ir

Bolshoylu, 2020-04-12:
Does anyone have a ready-made JSON configuration  for the remote control to the TV (for example Samsung), or an air conditioner, to understand the essence of how to do it. 
Thanks for your help

DanielSergienko, 2020-04-12:
`{"0":{"i":[{"c":"bAeBaFaAcBaAaCb","r":2}]},"1":{"i":[{"c":"bAeBaEbAcDaCb","r":2}]},"6":{"i":[{"c":"bAeBaEfAaFaA"}]},"7":{"i":[{"c":"bAeBaFeAbEaA"}]},"8":{"i":[{"c":"bAeBaHhE"}]},"9":{"i":[{"c":"bAeBaEaBeAbE"}]},"10":{"i":[{"c":"bAeBaEbAcAaBaCaA"}]},"11":{"i":[{"c":"bAeBaEcAaAbCaAaB"}]},"13":{"i":[{"c":"bAeBaEbEaBeA"}]},"17":{"i":[{"w":"E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAky$E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAky$E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAk"}]},"22":{"i":[{"c":"bAeBaEaAfAaF"}]},"23":{"i":[{"c":"bAeBaGhF"}]},"q0":[{"h":"192.168.100.99","r":0}],"q1":[{"h":"192.168.100.99"}],"i":[{"0":{"i":[{"c":"bAeBaEaCbCcBb","r":2}]},"n":"HDMI 1"},{"1":{"i":[{"c":"bAeBaEbEaBeA","r":2}]},"n":"Smart"}],"t":60} `
LG Remote with HAA

Bolshoylu, 2020-04-15:
> `{"0":{"i":[{"c":"bAeBaFaAcBaAaCb","r":2}]},"1":{"i":[{"c":"bAeBaEbAcDaCb","r":2}]},"6":{"i":[{"c":"bAeBaEfAaFaA"}]},"7":{"i":[{"c":"bAeBaFeAbEaA"}]},"8":{"i":[{"c":"bAeBaHhE"}]},"9":{"i":[{"c":"bAeBa [...]

this config does not work

xrust83, 2020-04-17:
> `{"0":{"i":[{"c":"bAeBaFaAcBaAaCb","r":2}]},"1":{"i":[{"c":"bAeBaEbAcDaCb","r":2}]},"6":{"i":[{"c":"bAeBaEfAaFaA"}]},"7":{"i":[{"c":"bAeBaFeAbEaA"}]},"8":{"i":[{"c":"bAeBaHhE"}]},"9":{"i":[{"c":"bAeBaEaBeAbE"}]},"10":{"i":[{"c":"bAeBaEbAcAaBaCaA"}]},"11":{"i":[{"c":"bAeBaEcAaAbCaAaB"}]},"13":{"i":[{"c":"bAeBaEbEaBeA"}]},"17":{"i":[{"w":"E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAky$E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAky$E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAk"}]},"22":{"i":[{"c":"bAeBaEaAfAaF"}]},"23":{"i":[{"c":"bAeBaGhF"}]},"q0":[{"h":"192.168.100.99","r":0}],"q1":[{"h":"192.168.100.99"}],"i":[{"0":{"i":[{"c":"bAeBaEaCbCcBb","r":2}]},"n":"HDMI 1"},{"1":{"i":[{"c":"bAeBaEbEaBeA","r":2}]},"n":"Smart"}],"t":60} `
> LG Remote с HAA

Привет. На каком Gpio сидит светодиод? Я не в одном коде не нахожу метку g и не нахожу описание по умолчанию

DanielSergienko, 2020-04-17:
> > `{"0":{"i":[{"c":"bAeBaFaAcBaAaCb","r":2}]},"1":{"i":[{"c":"bAeBaEbAcDaCb","r":2}]},"6":{"i":[{"c":"bAeBaEfAaFaA"}]},"7":{"i":[{"c":"bAeBaFeAbEaA"}]},"8":{"i":[{"c":"bAeBaHhE"}]},"9":{"i":[{"c":"bAeBaEaBeAbE"}]},"10":{"i":[{"c":"bAeBaEbAcAaBaCaA"}]},"11":{"i":[{"c":"bAeBaEcAaAbCaAaB"}]},"13":{"i":[{"c":"bAeBaEbEaBeA"}]},"17":{"i":[{"w":"E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAky$E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAky$E@AkB*AkAkAkB*AkAkAkB*AkAkAkAkAkAkAkAkAkAkAkAkAkB*AkAkAkB*AkAk"}]},"22":{"i":[{"c":"bAeBaEaAfAaF"}]},"23":{"i":[{"c":"bAeBaGhF"}]},"q0":[{"h":"192.168.100.99","r":0}],"q1":[{"h":"192.168.100.99"}],"i":[{"0":{"i":[{"c":"bAeBaEaCbCcBb","r":2}]},"n":"HDMI 1"},{"1":{"i":[{"c":"bAeBaEbEaBeA","r":2}]},"n":"Smart"}],"t":60} `
> > LG Remote с HAA
> 
> Привет. На каком Gpio сидит светодиод?
> Не нахожу метку g и по умолчанию

Там ее и нет. Это просто ветка аксессуара, я уж и не помню точно где она стоит к меня, но посмотри в вики в настройках «IR Actions” написано, по моему оно задается командой «I:”, но точно не g

xrust83, 2020-04-17:
> Там ее и нет. Это просто ветка аксессуара, я уж и не помню точно где она стоит к меня, но посмотри в вики в настройках «IR Actions” написано, по моему оно задается командой «I:”, но точно не g

Ок читаю. Я питаюсь запустить на wemos d1 mini. припаял Ir diode через транзистор как в примерах. Там было gpio-0 в примере так и впаял. Через камеру телефона вижу что он горит постоянно. В HomeKit появился телик, кнопка и 3 источника json взял от Samsung но команды пульта от LG https://github.com/RavenSystem/esp-homekit-devices/wiki/IR-Codes-Database Приложи скриншот как оно должно выглядеть в приложении Homekit пожалуйста

Bolshoylu, 2020-04-17:
> Там ее и нет. Это просто ветка аксессуара, я уж и не помню точно где она стоит к меня, но посмотри в вики в настройках «IR Actions” написано, по моему оно задается командой «I:”, но точно не g

а у меня конфиг не сохраняется выкидывает опять в настройки что может быть не так?

xrust83, 2020-04-17:
> Does anyone have a ready-made JSON configuration for the remote control to the TV (for example Samsung), or an air conditioner, to understand the essence of how to do it.
> Thanks for your help
samsung https://github.com/RavenSystem/esp-homekit-devices/issues/888

Bolshoylu, 2020-04-17:
> Does anyone have a ready-made JSON configuration for the remote control to the TV (for example Samsung), or an air conditioner, to understand the essence of how to do it.
> Thanks for your help
> samsung #888

это я пробовал
с этой конфигурацией пишет невозможно добавить устройство неправильный пароль

closed 2020-05-18 by stale[bot]