RavenSystem/esp-homekit-devices #884: Garage door opener Eachen GD-DC5

mientki, 2020-04-12:
I bought an interesting and well-made device for opening the garage door on AliExpress.  has 1 opening sensor.  all accessories to facilitate assembly.  has ewelink software.  I changed soft to HAA but I have a problem with JSON configuration.  I have no idea what are the gpio of the relay, sensor and LED.  did any of you have experience with this device and help?  this can be found on the internet under the name EACHEN GD-DC5.  Will this device work with HAA?  on the motherboard I see 2 connectors for flashing.  the other is labeled MCU.  what is this?
[image]
[image]
[image]

tonysprenk, 2020-04-12:
Looks like this device might communicate through serial commands with a secondary micro controller. This is not yet supported by HAA but @ravensystem has confirmed that it will be in an upcoming release.

mientki, 2020-04-12:
will you need to reprogram 2 systems separately?  how does such a device work?

tonysprenk, 2020-04-12:
No the esp chip sends serial commands to the second controller telling it what to do.

wwns, 2020-04-13:
[Here's a sample code](https://github.com/doleron/esp8266-1-channel-relay-board-with-mqtt)

`void setup() {
  Serial.begin(9600);
}

void loop() {
  byte close[] = {0xA0, 0x01, 0x01, 0xA2};
  Serial.write(close, sizeof(close)); 
  delay(2000); 
  
  byte open[] = {0xA0, 0x01, 0x00, 0xA1};
  Serial.write(open, sizeof(open)); 
  delay(2000);      
} `

closed 2020-05-13 by stale[bot]

Thiso6, 2020-10-29:
Hi, I successfully flashed tasmota by using the 4 pins (VDD, GND, RX, TX) of the WIFI part (not MCU). I'm now trying to configure it properly

Be2daErtin, 2020-12-21:
> Hi, I successfully flashed tasmota by using the 4 pins (VDD, GND, RX, TX) of the WIFI part (not MCU). I'm now trying to configure it properly

Hi, Did you manage to get it working?

Thiso6, 2020-12-21:
Yes it is working properly. The problem is that the microcontroller unit (MCU) is in conflict with the tasmota control. That caused it to keep toggling. The solution is to remove the MCU and the solder some bypass wires. It is running perfectly now. I followed this thread: https://community.home-assistant.io/t/gd-dc5-tasmota-ha/149245/22

gogreenpower, 2021-02-19:
> Hi, I successfully flashed tasmota by using the 4 pins (VDD, GND, RX, TX) of the WIFI part (not MCU). I'm now trying to configure it properly

Any idea how to access those 4 pins on this style of board?

[image]