RavenSystem/esp-homekit-devices #885: Window blinds problem

Spexx80, 2020-04-12:
Hi and sorry for My Bad english,

First of all Thanks for this great peace of software.
My problem is that i want to control 4 (or more) window blinds with one Nodemcu and some relays.
It’s possible for me to control one blind with the Nodemcu but not more. 
Is it possible to control more then one? If yes, with what kind of Json?

Thank you very much!

jangolen, 2020-04-15:
No problem, just copy same type 45 with another gpio.
I did it like this
{
  "c": { "l": 2 },
  "a": [{
    "t": 45,
    "o": 18,
    "c": 14,
    "f": 70,
    "0": { "r": [{ "g": 12, "v": 1 }, { "g": 13 }] },
    "1": { "r": [{ "g": 12 }, { "g": 13, "v": 1 }] },
    "2": { "r": [{ "g": 12 }, { "g": 13 }] },
    "3": { "r": [{ "g": 12, "i": 1 }, { "g": 13 }] },
    "4": { "r": [{ "g": 12 }, { "g": 13, "i": 1 }] },
    "f0": [{ "g": 0, "p": 1 }],
    "f1": [{ "g": 0, "t": 2 }]
  },{
    "t": 45,
    "o": 18,
    "c": 14,
    "f": 70,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 16 }] },
    "1": { "r": [{ "g": 15 }, { "g": 16, "v": 1 }] },
    "2": { "r": [{ "g": 15 }, { "g": 16 }] },
    "3": { "r": [{ "g": 15, "i": 1 }, { "g": 16 }] },
    "4": { "r": [{ "g": 15 }, { "g": 16, "i": 1 }] },
    "f0": [{ "g": 0 }],
    "f1": [{ "g": 0, "t": 2 }]
  }, {
    "t": 24,
    "g": 14
  }]
}

Spexx80, 2020-04-16:
Cool, thank you very much. Help‘d a lot!

closed 2020-05-16 by stale[bot]