RavenSystem/esp-homekit-devices #886: MagicHome RGBW + IR remote control and HAA

topzoe, 2020-04-14:
Hi,
I want flash a MagicHome RGBW+IR remote (like this:https://templates.blakadder.com/MagicHome_ESP-IR-B-v2_3.html)

Question: is it possible to keep the IR fonctionality with the IR remote control?
If yes, did someone have HAA Json exemple?
GPIO01 = IRrecv
GPIO05 = PWM2
GPIO12 = PWM1
GPIO13 = PWM3
GPIO15 = PWM4

Thank's a lot

toshibochan, 2020-04-14:
@topzoe You can use this https://github.com/maccoylton/esp-homekit-rgbw-strip
Go to wiki page, have instructions for install.

vniehues, 2020-04-14:
@toshibochan could you elaborate on how to use this? 
Im looking to do the exact same thing as @topzoe but i cant find the right config.json for my strip...

Thank you!

toshibochan, 2020-04-14:
Maccoylton firmware can chose the gpio pins using the EVE app and If your magic home led strip is RGB or RGB + IR turn off pure white on EVE app and if is RGBW or RGBW + IR you switch on pure white on EVE app.

vniehues, 2020-04-14:
nice, thank you!
Do you, by any chance, know a working config for RavenCore MagicHome RGBW? I'd rather not use the IR and not flash again...

topzoe, 2020-04-14:
> @topzoe You can use this https://github.com/maccoylton/esp-homekit-rgbw-strip
> Go to wiki page, have instructions for install. And you can use EVE app for setup the settings for your device. In the EVE app you can chose the gpio pins and if is RGB or RGBW.

Thank you a lot! It works great!!!  (Don't forget the code for add the accessory is "11111111")

closed 2020-04-14 by topzoe

vniehues, 2020-04-21:
@topzoe Can you tell me which pins you used for which color? Im about to flash it aswell :)

toshibochan, 2020-04-21:
@vniehues Don’t change the code. You can change the gpio using the EVE app.

vniehues, 2020-04-24:
@toshibochan okay thank you! 
I've seen you opened [this Issue](https://github.com/maccoylton/esp-homekit-rgbw-strip/issues/10)... Did you get that resolved?Because thats exactly my issue right now with raven. Everything else works as expected.

toshibochan, 2020-04-24:
You using RGBWW or RGB led strip? If you using RGBWW you switch ON pure white on eve app. If you use RGB led strip pure white have to be off and the light will be cold white.

vniehues, 2020-04-25:
@toshibochan Already found it. Thank you! :)