RavenSystem/esp-homekit-devices #888: Universal IR Device

TriderG75, 2020-04-15:
- Universal IR device
https://gearbest.app.link/7kIjGiVCz5 (Tested Type B and Type C versions)
https://a.aliexpress.com/_d7NKf18 (Not tested)
https://s.click.aliexpress.com/e/_d7qlxvI (Tested)
(Thanks to carminedelia, micheletorrini and PSB on Discord)

- This json is for managing a Samsung TV with an IR device (see links above).
IR commands can be found here : https://github.com/RavenSystem/esp-homekit-devices/wiki/IR-Codes-Database.
Also available Philips and LG IR commands.
Json has also an IP function to know if smart Tv is On or Off using ping and a dummy switch that allows to enter in setup mode by pressing it one time.

- Json:
{"c":{"t":14,"p":"J*J*AaAaAaDFAa"},"a":[{"t":60,"xa":0,"i":[{"n":"MENU","0":{"i":   [{"r":2,"c":"CeCfAaBcAaAbC"}]}},{"n":"SOURCE","0":{"i":[{"r":2,"c":"CeCeAhG"}]}},{"n":"MUTE","0":{"i":[{"r":2,"c":"CeCeDhD"}]}}],"0":{"i":[{"r":4,"c":"CeCfAfAaF"}]},"1":{"i":[{"r":4,"c":"CeCfAfAaF"}]},"6":{"i":[{"r":2,"c":"CeCjBaEbA"}]},"7":{"i":[{"r":2,"c":"CeCeAdBbDbA"}]},"8":{"i":[{"r":2,"c":"CeCeAaAbBbAaBbA"}]},"9":{"i":[{"r":2,"c":"CeCfAcBaAaCbA"}]},"10":{"i":[{"r":2,"c":"CeChAaBaCaAbA"}]},"11":{"i":[{"r":2,"c":"CeChBaAaCbAaA"}]},"13":{"i":[{"r":2,"c":"CeCeBaAaBcAaAbA"}]},"17":{"i":[{"r":2,"c":"CeCeEhC"}]},"22":{"i":[{"r":2,"c":"CeCeChE"}]},"23":{"i":[{"r":2,"c":"CeCeBaAfAaD"}]},"q0":[{"h":"192.168.1.100","r":0}],"q1":[{"h":"192.168.1.100","r":1}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

[image]

-  Gocomma Type B

[image]
[image]
[image]

- Cogomma Type C

[image]
[image]
[image]

- Avatto S07

[image]
[image]

closed 2020-04-15 by TriderG75

vijaydembla1, 2020-07-19:
Hi Trider,

I copied the exact same code , however i just get 3 buttons  (menu, source and mute) with one-off on my homekit. Any reason why i can’t see the remote or the other buttons?

Thanks

TriderG75, 2020-07-19:
> Hi Trider,
> 
> I copied the exact same code , however i just get 3 buttons (menu, source and mute) with one-off on my homekit. Any reason why i can’t see the remote or the other buttons?
> 
> Thanks
Hi, you can find the other commands in the remote in the control centre. When you use that remote for volume up/down you must use iPhone volume up/down side keys.

vijaydembla1, 2020-07-20:
Wow, i always thought i was technically smart but you proved me otherwise, i was looking in the remote app and not one in control centre.
What's the best way to include other tv's items in here, the question is to understand the way to caputre the code from the other remotes and incude them in my json

Thanks again for your help :-)