RavenSystem/esp-homekit-devices #889: Window blind with open sensor

jangolen, 2020-04-15:
Hi!
Is it possible add open sensor state to window covering action? Same as for garage door.
I have DIY motors with magnetic hall sensor on open state. It's very useful for proper full open position. With it no needed to set precision timing. 
Thanks.

jangolen, 2020-04-17:
Solved via 'Obstruction detected' then stop then message 'Obstruction removed'
Not a right, but worked.
{
  "c": { "l": 2, "o": 1 },
  "a": [{
    "t": 45,
    "o": 18,
    "c": 14,
    "f": 70,
    "0": { "r": [{ "g": 12, "v": 1 }, { "g": 13 }] },
    "1": { "r": [{ "g": 12 }, { "g": 13, "v": 1 }] },
    "2": { "r": [{ "g": 12 }, { "g": 13 }], **"m": [{"g": 1, "v": -2}] },**
    "3": { "r": [{ "g": 12, "i": 1 }, { "g": 13 }] },
    "4": { "r": [{ "g": 12 }, { "g": 13, "i": 1 }] },
    **"8": { "r": [{ "g": 13 }]},**
    "f0": [{ "g": 0, "p": 1 }],
    "f1": [{ "g": 0, "t": 2 }],
    **"f6": [{ "g": 4, "p": 1, "t": 0 }]**
  },{
    "t": 45,
    "o": 18,
    "c": 14,
    "f": 70,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 16 }] },
    "1": { "r": [{ "g": 15 }, { "g": 16, "v": 1 }] },
    "2": { "r": [{ "g": 15 }, { "g": 16 }], "m": [{"g": 2, "v": -2}] },
    "3": { "r": [{ "g": 15, "i": 1 }, { "g": 16 }] },
    "4": { "r": [{ "g": 15 }, { "g": 16, "i": 1 }] },
    "8": { "r": [{ "g": 16 }]},
    "f0": [{ "g": 0 }],
    "f1": [{ "g": 0, "t": 2 }],
    "f6": [{ "g": 5, "p": 1, "t": 0 }]
  }, {
    "t": 24,
    "g": 14
  }]
}

closed 2020-05-17 by stale[bot]

EgorMuhin, 2022-06-08:
> {
> "c": { "l": 2, "o": 1 },
> "a": [{
> "t": 45,
> "o": 18,
> "c": 14,
> "f": 70,
> "0": { "r": [{ "g": 12, "v": 1 }, { "g": 13 }] },
> "1": { "r": [{ "g": 12 }, { "g": 13, "v": 1 }] },
> "2": { "r": [{ "g": 12 }, { [...]

Good afternoon, please tell me how did you connect the magnetic sensor? thanks a lot in advance for the answer.