RavenSystem/esp-homekit-devices #890: HowTo? Get the right "Correction Value" for Window Covering (roller shutter)

Shimakuchi, 2020-04-16:
Hi, is there a step-by-step guide to get the right "[Correction Value](https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering#correction-value)"?

The wiki says **realPosition = coverPosition / (1 + ((100 - coverPosition) * correctionValue * 0.0002))**

Let's pretend I set the roller shutter via Siri to a desired value of 50% and its far more open or closed as it should be.

From my understanding this result (64% for example) is the "realPosition" then the coverPosition have to be the wanted 50%? Or is it the other way around? is the "realPosition" the the desired value?

So how do I get the **correctionValue** without guessing (try and error). 
**correctionValue = ...**

kind regards, Markus

----fyi----
Covering Type = roller shutter
Opening Time = 27
Closing Time = 23

Shimakuchi, 2020-05-11:
have nobody an idea 😅 I am wondering, is it just a dumb question ⁉️
**How do you tune in your settings?**

bye 👋🏻

Shimakuchi, 2020-06-16:
I am still wondering... as I have to update all my devices (#825 ) I would like to understand this settings so I can right dial in those during the update.

Kienz, 2020-06-24:
I'm also interested in an explanation - I don't understand the explanation in wiki. Thanks!

RavenSystem, 2020-07-05:
@Shimakuchi 
I updated Wiki: https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering#correction-value

`HomeKit Position = motorPosition / (1 + ((100 - motorPosition) * correctionValue * 0.0002))`

Shimakuchi, 2020-07-06:
thx @RavenSystem 
maybe [this](https://jsfiddle.net/pahup/5z7eLngy/43/) will help someone. Please feel free to check and improve the code as JS is nothing I usually deal with

Kienz, 2020-07-07:
@Shimakuchi Can you explain your code? Any real example? Did not unterstand your calculation.

Shimakuchi, 2020-07-07:
In short try the following value
HomeKit-Position = 50%
Motor-Position = 70%
Correction-Value = 67

---

Assume you set the height to 50% in HomeKit. To figure out the Motor-Position you could measure the hight of your window (1oo%) and figure out the Motor_height by utilise the rule of three. After that you know that the Motor-Position is 70% for example.

Put the 50% in the HomeKit-Position Textfield and the 70% in the Motor-Position Textfield. The code runs from 0 to 100% (correctionValue). If you in the range of possibility you get a result with some arrows. If there are no results, you maybe need to fix your [closing time](https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering#closing-time) first (or I did something wrong 🤪). 

In the example we use 50% HomeKit-Position, 70% Motor-Position. The CorrectionValue we are looking for should be 67%.

Since "49.928673323823105" matches the closest to the HomeKit Value of 50%.

Hope this helps

Shimakuchi, 2020-07-09:
@Kienz from what I see, you are far more experienced with javascript as I am. How about you write the code. In sum it is actually just a for loop which checks an Integer from 0-100 as the correction value and points out the correction value where the reuslt machtes closest to the expected homekit value. please let me know what you think. have a nice day

closed 2020-08-08 by stale[bot]