RavenSystem/esp-homekit-devices #891: Sonoff Basic TH16 as Heater with DS18B20

JayJayXPro, 2020-04-17:
Hi! What should I change in this code to work with the DS18B20 sensor? I want the measurement scale to be the maximum that the sensor can measure.

{"C": { "l" 13, "b": [{"g" 0, "t" 5}]}, "a": [{"t" 21, "s": 5 , "g" 14, "0": {"r": [{"g" 12}]}, "1": { "and" 0} "3": { "r": [{ "g" 12, "v" 1}]}; "5": { "and" 0}}]}

I want to connect TH16 to a water heater to turn it on and off automatically at a certain temperature.

Thanks!

wwns, 2020-04-22:
Measuring range: from -55 ° C to 125 ° C ,above 85 is not accurate.
Accuracy: +/- 0.5 ° C in the range -10 ° C to 85 ° C

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":21,"s":5,"g":14,"b":[],"n":3,"j":30,"w":1,"m":-10,"x":85},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[{"g":0,"t":1}],"s":0},{"t":24,"b":[],"n":3,"g":14}]} `

JayJayXPro, 2020-04-22:
Thank you very much! But I have one more question. I wanted to use my second TH16 + AM2301 to work with an electric heater (Range between ~ 10 ℃ and ~ 40 ℃.). The code provided on the "Devices Database" page does not work.
{"C": { "l" 13, "b": [{"g" 0, "t" 5}]}, "a": [{"t" 21, "s": 5 , "g" 14, "0": {"r": [{"g" 12}]}, "1": { "and" 0} "3": { "r": [{ "g" 12, "v" 1}]}; "5": { "and" 0}}]}

What should this code look like?

wwns, 2020-04-22:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":21,"s":5,"g":14,"b":[],"n":2,"j":10,"w":1,"m":10,"x":40},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[{"g":0,"t":1}],"s":0},{"t":24,"b":[],"n":2,"g":14,"j":10}]}

JayJayXPro, 2020-04-22:
There is still a problem with the temperature display 
:/
[image]
[image]

xrust83, 2020-04-22:
you can even insert a finger in the sensor.  but until drivers are added to your finger it won’t work.  in the list of sensors there is no am2301 / dhr21 but there are dht11 and dht22
[image]

https://github.com/RavenSystem/esp-homekit-devices/wiki/Thermostat

wwns, 2020-04-22:
This is the dh22 sensor, check sensor plug

JayJayXPro, 2020-04-23:
The device worked after changing the "n" from 2 to 4. Thanks to your code appears to me a thermostat, humidity and temperature sensors and a switch that I do not know what it is used for. What should I remove to get rid of this third switch? In my case I don't need it. Thanks so much for your help in this matter!
___________________________________
Urządzenie zadziałało po zmianie "n" z 2 na 4. Pdzięki twojemu kodowi pojawia mi się termostat, czujniki wilgoci oraz temperatury i przełącznik, który nie wiem do czego służy. Co powinienem usunąć aby pozbyć się tego trzeciego przełącznika? W moim przypadku jest mi zbędny. Dzięki wielkie za pomoc w tej sprawie!

wwns, 2020-04-23:
First string corrected in the second string added parameter ”d”

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":21,"s":5,"g":14,"b":[],"n":4,"j":10,"w":1,"m":10,"x":40},{"t":24,"b":[],"n":4,"g":14,"j":10}]} `

`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"a":0},"3":{"r":[{"g":12,"v":1}]},"5":{"a":0},"t":21,"s":5,"g":14,"b":[],"n":4,"j":10,"w":1,"m":10,"x":40,"d":5},{"t":24,"b":[],"n":4,"g":14,"j":10}]} `

Parameter „ d „ sets the temperature deadband [°C].
set it to 5 degrees, when the temperature drops, the heating will turn on.

closed 2020-05-20 by RavenSystem