RavenSystem/esp-homekit-devices #892: Esp-01 with a Switch for Philips Hue

tommasoconti98, 2020-04-17:
Hi to everyone, i was tring to install HAA on my esp-01 board, to control 2 Philips Hue lamps. Could someone help me in setting up the json? I can't understand how to set it up. In my board with Esp-01, there is a switch (on -> off, not a button) connected to the GPIO0, with a pullup resistor. What I would like is that when the state of the switch changes (from on to off, or vice versa) the state of 2 philips bulbs changes, based on their state. So if they were on and the state of the switch changes, (again from on to off or vice versa) they will turn off. How to do? Just configure it as a switch (which I don't know how to do it anyway) and then from the homekit set that when its status changes, that of the two lights also changes, or configure it that when it changes its status it sends a direct command to the two lights (which also this I don't know how to do it)? Thank you

bj-tracer, 2020-04-28:
Hello!
You want to conrol esp-01 board remotely from Home app only? Or you need an button on the device?
Are the Hue Lamps connected to the GPIO0 of esp-01?

Please check this string:
```
{
  "c": { "l": 13, "b": [{ "g": 0, "t": 5 }] },
  "a": [{
    "t": 1,
     "s": 5,
    "0": { "r": [{ "g": 0 }] },
    "1": { "r": [{ "g": 0, "v": 1 }] },
    }]
}
```
It's a simple switch witout real buttons. Controls only from Home app.

tommasoconti98, 2020-04-28:
I probably didn't explain myself well. I have 2 Philips Hue lamps, which I
want to switch on and off with a switch, but which does not remove the
power from the lamps themselves. In fact, I connected a switch to the GPIO0
input of an ESP-01, and I would like that when the state of the switch
changes, the command is sent to the two bulbs. Fimrware that use direct
integration with Hue Hab has a really long delay, and I thought of
integrating it directly through the Home app. Do you think it is possible?
Thanks

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

peros550, 2020-04-28:
It should be possible. Someone with good skills on json will build one for you. The way I see it, all you need is a json for a simple switch connected at the GPIO of your choice. 

But in order for you to use your Hue bulbs, they must be Enabled in the HomeKit and you need to create some Automations using a HomeKit hub (Apple TV 4th generation or higher, iPad , HomePod ) 

One automation for turning on the Hue bulbs when the switch is turned on
and one other automation (rule ) for turning off the Hue when the switch is turned off

If you use push buttons, the automations would need to be created using another app which takes into account conditions. For example : 

1st automation: 
If I press the button AND the bulbs are On ->> turn them off 

2nd automation:

If I press the button AND the bulbs are Off->> turn them on

tommasoconti98, 2020-04-28:
And its not possible to do things like the ones of the buttons, but
istead of make some rules with automations of Homekit hub, do the "math" in
the arduino (instead of button pressed, trigger when the input change
state)?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bj-tracer, 2020-04-29:
Ok.
If you connected esp-01 to the proprietary hue switch you can simulate press button on it only.
You can't check the state of hue bulbs at HK.
Normally HK switches hasn't momentary state. It's work like a toggle - if the virtual switch is on then real gpio have permanent state like 1 (it depends on the settings and can be 0).
So you need to add delay for automatically switch off esp-switch. Check the wiki for "Inching".

```
{
  "c": { "l": 13, "b": [{ "g": 0, "t": 5 }] },
  "a": [{
    "t": 1,
     "s": 5,
    "0": { "r": [{ "g": 0 }] },
    "1": { "r": [{ "g": 0, "v": 1,"i":0.5 }] },
    }]
}
```

Test this json.
If you switch on it at Home app it shoud be automatically HK toggle off after 0.5 sec.

closed 2020-05-20 by RavenSystem

pongista, 2022-01-21:
@RavenSystem  Hi Joseè, I need your help if is possible, I install HAA on ESP01 with relays 4.0, I need to compile a Jason to have just a switch with 1 second inching ( I will use to switch on a mini pc does not have a WOL) can you help me please? thanks in advance