RavenSystem/esp-homekit-devices #893: Sonoff L1 RGB , HomeKit pls help

msdem98, 2020-04-17:
Good day,
Is it possible to add this accessory to HomeKit, I am already flashed controller but after adding to HomeKit no any reaction of trying change color or switch on/off. I am noob, is the problem in wrong Json code and if yes, can somebody give me right code. 
[image]
[image]
[image]
[image]

msdem98, 2020-04-18:
[image]

tonysprenk, 2020-04-18:
This device uses serial commands to communicate with a secondary MCU this is not (yet) supported by HAA. @RavenSystem is working hard to have it supported in an upcoming update.

closed 2020-05-18 by stale[bot]

mwouda, 2021-04-11:
Did anyone have any success with getting the L1 controler to work? 

Can you share the JSON?
I'm not having any success

thx.

DrDaKen, 2021-05-18:
Hi, i have the same problem, cannot figure out the json strings.
please help to share!
regards,

mwouda, 2021-05-21:
> This device uses serial commands to communicate with a secondary MCU this is not (yet) supported by HAA. @RavenSystem is working hard to have it supported in an upcoming update.

Any Progress with  communicate with a secondary MCU

nikogeek, 2021-07-13:
any new info about it?