RavenSystem/esp-homekit-devices #894: Sonoff 4ch pro r2 hangs for 2 minutes and restarts

KonradTkaczyk, 2020-04-17:
First of all I will say that I am not sure this is an issue as use case is rather on the edge. So by constantly turning on and off switches after 15-20 seconds sonoff 4channel got freezed and then autorestarts. I am wondering that maybe this is protection for the latches but I am not sure... I have used configuration code from Raven Wiki which is here -> https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database?fbclid=IwAR2CGrkO098KRWzJzFuU0ek8KagjlUQE7LR1tZzaZUQh8KDa0i1h4atsKDQ
For device:
Sonoff 4CH

Code is this one:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}

I did not modified it at all. I compared with some different codes and noted down that others have "s": 0 which stands probably for state of the latch after reset...

Video of the issue can be watched here: https://youtu.be/H8dklmEtnjQ

xrust83, 2020-04-18:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#entering-setup-mode

KonradTkaczyk, 2020-04-19:
Its not a bug its a feature :)as mentioned above this is setup mode. I am closing the ticket. Thanks for help:)

closed 2020-04-19 by KonradTkaczyk