RavenSystem/esp-homekit-devices #896: Can entering the setup mode be solved more elegant?

morphix9, 2020-04-19:
What I really "hate" right now is triggering of the setup mode. I'm using Shelly 1's mostly, which don't have a physical button and turning off those shelly's and then triggering the switch 8 times, is really cumbersome if you have many devices throughout the house. Is there a way to trigger the setup mode over HTTP? Is it possible to run a simple web server all the time on the device (while working with HomeKit)? This would really make our life easier :)

pilot1981, 2020-04-19:
You can create a dummy switch to enter in setup mode when you want!

PsbDxb, 2020-04-19:
add this code along
{"i":0.5,"1":{"s":[{"a":1}]}}

it will add a dummy switch in homekit, once triggered it will put the device into setup mode.

example of a sonoff basic  with dummy switch code 
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

morphix9, 2020-04-20:
Thanks, a dummy switch would be great. This is clearly missing in the wiki.

morphix9, 2020-04-23:
Is it possible that by adding an extra switch to trigger setup mode, Siri gets confused and thinks I have 2 devices with the same name (although the switch is named HAA-XXXXX)?

pilot1981, 2020-04-23:
yes, one is for setup mode and the other is for real relay.
I suggest you to rename dummy switch and move it to dummy room.
To find the right switch you have to try!

morphix9, 2020-04-23:
Exactly what I've done. One is "Garage Door" in room "Garage", other is "HAA-XXXXX" in room "Setup mode". Looks like Siri gets confused because it's the same device behind those two accessories.

pilot1981, 2020-04-23:
Which type of problem you have?
I haven’t any similar issue.

Please share some screenshoot of your configuration.

morphix9, 2020-04-23:
[image]
[image]

Kienz, 2020-04-23:
Never had this issue. I have some devices with multiple switches (Setup-Mode, OTA-Update, Light, Switch). I renamed all accessories (eg. "... SetupMode", "... OTAUpdate", "... Light", "... Switch"). And Siri works without any issues.

pilot1981, 2020-04-23:
You can do an example of Siri problem?

morphix9, 2020-04-24:
[image]

pilot1981, 2020-04-24:
Have you already tried to rename dummy switch?

closed 2020-05-19 by morphix9