RavenSystem/esp-homekit-devices #897: Simple Relay Switch

McVareck, 2020-04-19:
I'm new in this field and I'm trying to create a switch connected to the intercom to open the door downstair. I flash RavenSystem HAA 1.10 in a ESP8266 and I managed to connect it with iPad and iPhone. I can see the button in the Home app, I can turn on and off but nothing happens to the relay, it doesn't work. I tried three different JSON configuration, Sonoff Basic/SV/Mini/TH/POW, Stateless Switch and Lock Mechanism. How can I do to make it works?

tonysprenk, 2020-04-19:
Hi @McVareck , welcome! Can you please post your json? And which gpio is the relay connected to?

McVareck, 2020-04-19:
Hi, I just try this Stateless Button for ESP-01 {"c":{"b":[{"g":0,"t":5}]},"a":[{"t":3,"f0":[{"g":0}],"f1":[{"g":0,"t":2}],"f2":[{"g":0,"t":3}]}]} and it doesn't work. Maybe GPIO0. I don't know. I attach you some photos. I'm powering relay with 5V of USB/TTL adaptor. Relay works when I connect it to computer or when I press RST bottom so I think it's ok.
[image]
[image]
[image]
[image]
[image]

tonysprenk, 2020-04-19:
Your json at the moment is configured for gpio0 which is unlikely to be connected to the relay as it is normally used to put the esp in flash mode. Try to configure your json with gpio 1, 2 or 3.

tonysprenk, 2020-04-19:
Hmm scratch that. Looking closer at your pictures it looks like it’s indeed gpio0 that is connected to the relay. I’m not sure why it wouldn’t work. Maybe someone with more JSON knowledge can help. Also try asking your question in the json config channel on the discord.

McVareck, 2020-04-19:
I should try but I know how to do it. I tried to change all the "g"0 in "g"2 but I know if it have any sense. Anyway I tried with this JSON {{“c”:{“b”:[{“g”:2,”t”:5}]},”a”:[{“t”:3,”f0”:[{“g”:2}],”f1”:[{“g”:2,”t”:2}],”f2”:[{“g”:2,”t”:3}]}]} but iPhone can't see it

wwns, 2020-04-22:
[Answer yourself, here you have everything](https://github.com/IOT-MCU/ESP-01S-Relay-v4.0/blob/master/ESP-01S%20Relay%20v4.0.pdf)

`{"":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":2}]},"1":{"r":[{"g":2,"v":1}]},"b":[{"g":0}],"s":0},{"0":{"r":[{"g":2,"v":0},{"g":2,"v":1}]},"t":3,"b":[],"s":0},{"0":{"r":[{"g":2,"v":1}]},"1":{"r":[{"g":2,"v":0}]},"t":4,"b":[],"s":0}]} `

bj-tracer, 2020-04-23:
Hello!
This pcb containt gpio0 only. The gpio2 is not traced.
You need to solder button to the gpio2 pin and gnd.
Then add this button to the json string as normal input instead stateless button.
Also you can use gpio3 - it's RX pin on your pcb.

You need to make an automation at the Home app if you want to use stateless button yet.
Try to add single press action for relay switch on and double or long press for switch off.
For the single press on/off you need to use Eve app - it can do more powerfull actions.

closed 2020-05-20 by RavenSystem