RavenSystem/esp-homekit-devices #898: SONOFF SV Relay Not Triggering.

the1maximus, 2020-04-20:


wwns, 2020-04-22:
give json string

the1maximus, 2020-04-22:
@wwns I found the issue to be with the 3rd party configurator. For whatever reason it was not pushing the digital output to GPIO 12. 

I went back to the beginning and used the code from the device data base and started modifying that and have not had the problem.

closed 2020-04-22 by the1maximus

mestregroda, 2020-10-27:
@the1maximus I am stuck with the same issue, but I am using sonoff mini. Could you post the json the was sucessfull for your setup?

the1maximus, 2020-10-27:
> @the1maximus I am stuck with the same issue, but I am using sonoff mini. Could you post the json the was sucessfull for your setup?

This is the only code I could get to work. The relay will trigger when power cycled. in the home app, it will either show on or off, but each trigger is actually a press of a momentary button. 

{
	"c": {
		"l": 13,
		"b": [{
			"g": 0,
			"t": 5
		}]
	},
	"a": [{
		"0": {
			"r": [{
				"g": 12,
				"v": 1,
				"i": 0.5
			}]
		},
		"1": {
			"a": 0
		},
		"b": [{
			"g": 0
		}]
	}]
}

RavenSystem, 2020-10-27:
To avoid initial relay trigger:
`{"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},"a":[{"xa":0,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"b":[{"g":0}]}]}`

the1maximus, 2020-10-27:
> To avoid initial relay trigger:
> `{"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},"a":[{"xa":0,"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"b":[{"g":0}]}]}`

Thanks for the reply, I’ll try this out and let you know.