RavenSystem/esp-homekit-devices #899: [Configuration]

vniehues, 2020-04-21:
I successfully flashed some devices using this firmware.
One device is a RGBW LED-Controller. RGB are displayed correctly however i'm having some troubles when i try to only activate the white LEDs through Homekit.

Is there a way to add a "deadzone" where not matter where i am in the color picker, it defaults to pure white?

Or are there any other workarounds?

Thanks in advance! :)

wwns, 2020-04-22:
2-channels selectable temperature color: declare "v" for cool white and "w" for warm white. 
RGB: declare "r", "g" and "v" for Red, Green and Blue colors. 
RGBW: declare "r", "g", "v" and "w" for Red, Green, Blue and White colors.

vniehues, 2020-04-22:
@wwns right, i got that working as i desribed above..
the issue is, that i cant address **only** the white LEDs... the RGB ones also turn to a "white-ish color" but i want the off. 
not matter, what color or "white color" i select in the homekit color picker, it doesnt work.

RavenSystem, 2020-05-20:
Sorry, there is no way to turn off RGB part when a white color is selected.

closed 2020-05-20 by RavenSystem

toshibochan, 2020-08-01:
Hi @RavenSystem you can look this https://github.com/maccoylton/esp-homekit-rgbw-strip.  Maccoylton Magichome version have pure white opinion in he’s code. I was using he’s code but it keep loosing connection to homekit. So I start using yours today. But I love to have pure white if possible. Can you take look he’s version and try to and in yours magichome version. Please
Thank you for sharing amazing work with us.