RavenSystem/esp-homekit-devices #900: Reversed Digital Output

the1maximus, 2020-04-22:
When adding an inching interval to the digital outputs, I find that within the Home App, the affects are reversed on the switch. Basically turning off the switch in the Home App, triggers the physical switch to turn on. Turning on the switch in the Home App, turns the physical switch off. 

{
	"c": {
		"l": 13,
		"b": [{
			"g": 0,
			"t": 5
		}]
	},
	"a": [{
		"0": {
			"r": [{
				"g": 12,
				"i": 0.5
			}]
		},
		"1": {
			"r": [{
				"g": 12,
				"v": 1,
				"i": 0.5
			}]
		},
		"b": [{
			"g": 0
		}]
	}]
}

the1maximus, 2020-04-22:
To achieve the affects I want here is the code I found that works. 

{
	"c": {
		"l": 13,
		"b": [{
			"g": 0,
			"t": 5
		}]
	},
	"a": [{
		"0": {
			"r": [{
				"g": 12,
				"v": 1,
				"i": 0.5
			}]
		},
		"1": {
			"a": 0
		},
		"b": [{
			"g": 0
		}]
	}]
}

This code however brings on a new problem.  When the device reboots, It triggers GPIO 12. 

I am basically using this to trigger a momentary switch on my garage door remote, to cycle the light on the motor. If the power for whatever reason goes out, when it comes back, this will triggers my light to come on.

the1maximus, 2020-04-24:
@bj-tracer Thank you for pointing that out. Let me explain what exactly it’s being used for, just so we are both on the same page. My Sonoff SVs are set up for “Dry Contact” mode. On one side the +/- are connected together, on the other side 2 wires are connected to a momentary button on my garage door remote. Pressing the button on the remote will trigger my light to turn on. Triggering the relay on the Sonoff jumps the terminals on the button as if it was a button press. To achieve the effect of a “button press” each time, I have to have inching on both the On trigger and Off trigger. 

  I did try alternatives to the codes I posted above. The 2nd code is the only code I could get to work properly, for the most part. 

In the first code, If I remove the inching, The relay will trigger properly, when Turned On but Turning it Off, does not trigger the relay again. 

So if I trigger my Sonoff, the light comes on, but turning the light off does not cause the button press, so the light stays on. 

I would like that, each time the physical button or the digital button in the Home App is triggered, it would act as a Momentary button. In affect almost like a stateless button, but instead of just triggering a different device, I wanted it to trigger itself. 

The only resolution I could find was the 2nd code. 
By inverting the trigger, this causes the relay to Trigger each time. This achieves the desired effect, but the negative is that on reboot it causes it to trigger as well.

bj-tracer, 2020-04-24:
You need to remove "off" action at 1st script I think.
Because you have inching for "on" action and your relay automatically switch off after 0.5 sec.

the1maximus, 2020-04-24:
Got it, Thanks again, I haven’t tried that ... I will try it and report back.

the1maximus, 2020-04-26:
@bj-tracer Thank you again for the suggestion but unfortunately no success.  With the code setup like this, the on action would trigger the relay, and the digital switch in the Home App would stay on. Toggling the switch “off” in the Home App has causes no action. I need both actions to cause the relay to trigger.

Because of the inching on both actions, both need to have a value of 1, because both need to trigger the relay to the on position to close the circuit for the momentary action. The adverse action this is causing is that on reboot, the device always toggles the trigger. I have also tried adding a initial state “s” action to 0 but that does nothing. 

{
	"c": {
		"l": 13,
		"b": [{
			"g": 0,
			"t": 5
		}]
	},
	"a": [{
		"1": {
			"r": [{
				"g": 12,
				"v": 1,
				"i": 0.5
			}]
		},
		"b": [{
			"g": 0
		}]
	}]
}

closed 2020-05-26 by stale[bot]