RavenSystem/esp-homekit-devices #902: PWM Frequency Set Does/Might Not Work

mriksman, 2020-04-23:
FYI. 

I'm working with the `multipwm `library of esp-open-rtos at the moment, porting it to ESP8266 RTOS SDK, and noticed the `multipwm_set_freq()` doesn't actually do anything.
https://github.com/RavenSystem/esp-homekit-devices/blob/master/devices/HAA/main.c#L4434

The actual frequency/period is set in the header with 
`#define MULTIPWM_MAX_PERIOD UINT16_MAX`

`multipwm_set_freq()` will cause the divider to be set to `TIMER_CLKDIV_16 `(see https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/timer.h)

So the frequency of the PWM calculates to about 76Hz. If a user selects a frequency > 100kHz, the divider will be `TIMER_CLKDIV_1 `so the PWM frequency will be 1.22kHz.
This is assuming the clock is running at 80MHz.

Is anyone able to confirm this?

RavenSystem, 2020-05-20:
I use esp-open-rtos and multipwm_set_freq() works perfectly. I use 1KHz for max value.

closed 2020-05-20 by RavenSystem

mriksman, 2020-05-20:
I spoke with the developer of `multipwm` and he confirmed he never implemented `multipwm_set_freq` properly. Record the LED with a slow-mo camera (like an iPhone) and you'll see it flashing.

blaxwan01, 2020-08-24:
@RavenSystem any solution? any value of "q"  in JSON  becomes 76Hz on PWM, I tested a lot of tries with oscilloscope