RavenSystem/esp-homekit-devices #904: Sonoff S26 and Setup Mode

paolocrybaby, 2020-04-23:
Hello Everyone,
I open a new issue about a different device but I still have the same problem: entering setup mode.

I just flashed a sonoff S26 using this JSON (From The collection):

> {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":2,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

(I am not expert of JSONs, but I understood this would go into setup mode if I push and hold the button for 8 seconds)

After pairing with Home, I tried to enter setup mode before unsoldering, in order to check that I can re-setup in case I will change LAN Settings, without reflashing from the beginning... no way I could achieve that.

I tried 4 options (with duponts still soldered and connected to the USB-TTL):

1. Connecting to usb to power it up, pressing 8 secs - doesn’t work

2. Pressing before connecting to USB, holding 8 secs... goes into flash mode, of course - doesn’t work

3. Connecting to power outlet (Without remounting the S26) and press/hold 8 secs - doesn’t work

4. Pressing before connecting to power outlet (Still S26 opened and soldered) and holding 8 secs - doesn’t work

Am I using a wrong JSON?

wwns, 2020-04-23:
Find the ip address in the router's dhcp settings. 
Press and hold on the device button while the LED is blink, release the button.

paolocrybaby, 2020-04-25:
Hi wwns thanks for helping. 
How can I identify the sonoff IP, considering that I have 15 devices connected to the router? And what do I need it for? Doesn’t it create the AP HAA-XXXXXX like during the first setup?

Also, one more question: let’s assume I move to another house, without being able to track the sonoff IP (because it is not connected to the WiFi anymore)... how can I get into setup mode and set up the new connection without re-flashing from scratch?

wwns, 2020-04-25:
only remove wifi settings
the device will be visible as at the first configuration.

blackwalkerhd, 2020-05-01:
Same probleme here with 15 sonoff basic, 2 sonoff 2 ch, 1 sonoff 4 ch
The sonoff basic are almost all the latest model and some (3) are the previous model
The sonoff 4 ch its a r2 model

For example, with sonoff basic i used the string from device database:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

used the string from device database for all my sonoff device.
I have never been able to enter the setup panel. Every time I have to disconnect the sonoff, update it and reassemble it.
Ok I can use the auto update, but I'd rather decide when to update and whether to do it

I have always used windows for flash, but I did the test sometimes with ubuntu and always the same thing.

It's not a serious problem but I'd like to use this feature!

paolocrybaby, 2020-05-02:
Hi, 
I just tried also with a Sonoff T3EU 2 gang with this JSON:
`{ "c": { "l": 13, "b": [ { "g": 0, "t": 5 } ] }, "a": [ { "0": { "r": [ { "g": 12 } ] }, "1": { "r": [ { "g": 12, "v": 1 } ] }, "b": [ { "g": 0 } ] }, { "0": { "r": [ { "g": 5 } ] }, "1": { "r": [ { "g": 5, "v": 1 } ] }, "b": [ { "g": 9 } ] } ] }`

It works in homekit (although it shows up as two different switches instead of one switch with two toggles), but if I hold toggle 1, it beeps once after 5 secs, and twice after around 8 secs.
Same if I hold toggle 2 or both toggles together.

Sometimes I could see the HAA-XXXXXX AP in my network list but, if I try to access, it gives me "unable to join". No way I can join the AP.

Just to clarify, since I haven't been able once to get into setup mode after the initial setup: when I try to get into setup mode holding the toggle, what will I see?
Automatically the HAA-XXXXXX AP in my network list, where I can then select the wifi and JSON?
Or should I get into the device via browser with some sort of IP/MAC address? (which means I should know what it is... and, in my router config page, how can I differentiate the device I want to setup among 10 devices I have at home?)

Thanks for clarifying, this "setup mode" is still really foggy

blackwalkerhd, 2020-05-02:
me too, when i try to enter into setup mode i see always the HAA-XXXXXX AP in my network list, for a short period and i quick try to enter but no way...wont let me in!!! Same wifi... i try everythng...

RavenSystem, 2020-05-02:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

blackwalkerhd, 2020-05-02:
> https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

thx for this info, i dint see it before, i'll try later! thx again :)

paolocrybaby, 2020-05-03:
Thanks RavenSystem. I actually managed to enter the config page (using the http://HAA-XXXXXX) when the wifi network was active (if disactivated, it works joining the AP). Thanks for clarifying.

One question: I saw an option to toggle about "update all OTA now".
During the configuration I always toggle the OTA auto update: does this mean that the firmware gets updated automatically? or should I go through setup mode every time?

RavenSystem, 2020-05-03:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#automatic-ota-updates

paolocrybaby, 2020-05-03:
Ok, nothing. I need to take a step back. No way I could change any parameter while in setup mode.

Yes, I managed to get into the configuration page of the setup mode, and I tried:
- to simply change the wireless network - no connection to the wifi specified
- to cancel all wifi settings - I expected it to get directly into setup mode but instead I need to trigger it anyway
- to reset Homekit ID - no changes, I still see the JSON when re-triggering setup mode
- to change the JSON - when I get into setup mode, I can see the initial JSON but not the new one...

Also, it keeps setting on-off one of the toggles by itself. :O
Any idea how to fix this TX EU 2 gang? (avoiding re-flashing from scratch, I could do that but I would like to test the easier way when changing WiFi)

closed 2020-06-02 by stale[bot]