RavenSystem/esp-homekit-devices #906: OTA failing

morphix9, 2020-04-25:
@RavenSystem any idea? Why is it trying to download 3.0.1?

HTTP returns 302 for Current ROM is 0
Compare
3.0.1 with 2.2.3 = 1
RavenSystem/haa_ota@version:"3.0.1"
Current ROM is 0
Running BOOT
Get hash
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.1/main.bin.sig HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=e505 DNS IP: 140.82.118.4 local..OK remote..OK
sent OK
HTTP returns 404
Get file
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/3.0.1/main.bin HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=e506 DNS IP: 140.82.118.4 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 404

RavenSystem, 2020-04-25:
It seems that you are using an old haaboot.bin file.

morphix9, 2020-04-25:
You're right! Thanks man!

closed 2020-04-25 by morphix9