RavenSystem/esp-homekit-devices #910: Not be found in Apple Home

DragoWob, 2020-04-26:
The Flash with the RavenCore2 works.

The HAA-XXXXXX Device create a Wifi.
After put in the json at it works 2 times the device restart.
Joins sucessfuly in the Wifi of all other devices.

But i cant find it in the apple home app.
What could i do?

restart and reflash several times.

I start to become crazy

RavenSystem, 2020-04-26:
Which JSON?

DragoWob, 2020-04-26:
{
  "c": { "l": 0, "b": [{ "g": 2, "p": 0, "t": 5 }] },
  "a": [{
    "t": 45,
    "o": 26,
    "c": 24,
    "f": 70,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 4 }] },
    "1": { "r": [{ "g": 15 }, { "g": 4, "v": 1 }] },
    "2": { "r": [{ "g": 15 }, { "g": 4 }] },
    "3": { "r": [{ "g": 15, "i": 1 }, { "g": 4 }] },
    "4": { "r": [{ "g": 15 }, { "g": 4, "i": 1 }] },
    "f0": [{ "g": 5, "p": 0 }],
    "f1": [{ "g": 13, "p": 0 }],
    "f2": [{ "g": 5, "t": 0 }, { "g": 13, "t": 0 }]
  }]
}

It Worms two times perfektly.

But now for pitty Not.
The Same bin the method the Same Laptop the Same Json.

I am wondering the device get in the wifi with „unknown“

RavenSystem, 2020-04-26:
I see that you are disabling internal pullup resistors. Are you using external pullup resistors?

DragoWob, 2020-04-26:
I think it is a misstake

Now i remove ma Existing two from Apple Home.
The First I found very fast
But the second Not 

Is there a Problem sing zwo times the Same id?
The Manuel Switch works

xfdr0805, 2020-04-27:
I have the same problem! can not be found device

kenji922, 2020-04-27:
I got same problem here. It was ok few days ago but now when you remove it, It will not be found again.

DragoWob, 2020-04-27:
Any idea of an Solution?
I complete Erase Apple HomeKit and install Every device new.
But the shelly still Not found 

Also a Philipps light is Not found any More

kompak23, 2020-04-27:
I have had the same problem. But in my case, the problem was that I falshed HAAinstaller and since few days, it is inpossible to connect devices with OTA server - so it cannot download files. That is why it is called in router "unknown device". 
Via Serial it sends informations that it attpempt to connect but there is error 404. And again... and again... 
Do you know what can be the reason? 
Solution:  flash newest verision directly :)

DragoWob, 2020-04-27:
I disable the ota in setup Mode.
Two times it works 5 other devices Not.

For pitty i am in trouble to make one bin to flash. So i was very happy to find one merged bin from the needed And used it.

DragoWob, 2020-04-27:
Maybe someone could help me and create for me a Single bin for a shelly 2.5 Window shutter That i could Flash on the shellys and try out

tonysprenk, 2020-04-27:
@DragoWob why do you need a single bin?

pmylius, 2020-04-28:
a single bin was needed in order to flash the devices in one go using NodeMCU.

However, neither using the newest OTA- nor w/o-OTA-release proved to be helpful.

We encountered several instances, where new devices could not be added to Homekit.

The device opens a Wifi just fine, can be configured and logged into the respective Home-Wifi, but isn‘t found when putting in the HomeKit-Code

tonysprenk, 2020-04-28:
I’ve noticed that when there are devices that are already added to the home app but have no response for whatever reason then the home app won’t find the new device. First I resolve or remove the no response devices then I can add the new device no problem. Might be worth a shot.

tonysprenk, 2020-04-28:
Easiest way to combine all bin files can be found [here](https://github.com/ct-Open-Source/tuya-convert/wiki/Flash-a-multipart-binary).

DragoWob, 2020-04-28:
I found a was to combine the bin.

@pmylius for Pitty Not responding decices are Not included in home. I remove already all of them. But i dont find the New one.

I Used the newest versions.

pmylius, 2020-04-28:
thank you @tonysprenk

pmylius, 2020-04-28:
@RavenSystem tried several JSONs but the problem to find new devices in HomeKit still remains - any ideas?

RavenSystem, 2020-04-28:
Logs, please.

pmylius, 2020-04-28:
do you mind to provide a quick HowTo for Logs for Windows?

pmylius, 2020-04-29:
Issue resolved - tried again with newly flashed shellies, which are now found by HomeKit.

Maybe it was a temporary glitch - but thank you for your help @all

closed 2020-05-29 by stale[bot]

Quimeric, 2021-04-17:
@RavenSystem i'm having the same problem.

First the Garage Door was behaving stranger and i bought the HAA Manager to try to update.

Unsuccessfully due the old firmware, I've reflashed with 5.0.16 firmware through NodeMCU (fullhaaboot.bin) and restored the old JSON which works for some time (a garage door with one sensor and when i click the garage door cease to open/close and stays, due  the fact that it a horizontal door e is useful to open  little bit)

At this time i can't find anymore the Sonoff SV, even flashing back a old firmware or using others JSON form device database (tried with a simple button and a garage door).

To extend my test i've tried with a Sonoff Mini (no flashing) and after i remove from Home App i was never able to find it anymore.

I simply can't re-add the devices because they never shows up.