RavenSystem/esp-homekit-devices #913: HAA - Automate sprinkler

claire078, 2020-04-29:
Hi,

I’d like to automate my watering system but I cannot a time schedule because I can not select sprinkler in Home App. HAA 2.1
Is there a solution?

macjl, 2020-04-29:
Use Eve to create a scene that activate Sprinkler (and you can also choose the duration with Eve if you want). Then you can launch this scene with a schedule in Home.

claire078, 2020-04-30:
Yes, this can work. 👍
But it works with Ravencore, HAA will support it as well?

pilot1981, 2020-04-30:
Yes, it work also for HAA
[image]
[image]
[image]

claire078, 2020-04-30:
I meant HAA automation works without EVE app.

pilot1981, 2020-04-30:
Yes

claire078, 2020-05-22:
OK, again: could we will use watering accessories in automate rules without set a scene in Eve app?

RavenSystem, 2020-05-22:
@claire078 Ask Apple.

claire078, 2020-05-22:
> @claire078 Ask Apple.

You say, it does not work cause apple disabled these functions?
But why would they did that?
A sprinkler accessory without schedule option? That is make no sense...

closed 2020-06-21 by stale[bot]