RavenSystem/esp-homekit-devices #915: Feature Request: receiving UART commands

tonysprenk, 2020-04-30:
Now that sending UART commands is supported by HAA it would be really nice to be able to receive and process UART commands from the secondary MCU. This is especially helpful when a device has more than one way of controlling it. For example my curtains can now be controlled with the home app but they also have a feature where if you pull the curtain slightly to open or close the motor will take over automatically. When that happens the MCU sends a packet to tell the ESP that the curtain is opening or closing. But because HAA doesn’t support receiving UART communications, only sending, the status of the curtain stays the same in the home app even though the physical state has changed. This can also be useful for obstruction detection. 

Tasmota handles this in their tuymcu templates so it should be possible. 

If anybody has any more info or ideas on this feel free to comment. 

More info [here](https://tasmota.github.io/docs/TuyaMCU/#tuya-protocols).

HomeACcessoryKid, 2020-05-03:
If your model is this aqara motor then I suggest you look at AQARA-ZNCLDJ11LM <https://github.com/HomeACcessoryKid/AQARA-ZNCLDJ11LM>

tonysprenk, 2020-05-03:
It’s not the same motor but it does have the same rj45 connector function. Can i use the rf remote with your solution and the status will be updated in the home app?

HomeACcessoryKid, 2020-05-03:
All depends on what the protocol is. The connector is RJ11, not RJ45 BTW

wayn3, 2020-05-08:
Would be awesome to have this feature, as an example I have a Tuya based fan controller that will receive uart commands from an rf remote via the tuyamcu

When the rf remote switches the fan on/ff, alters the speed or switches the light on/off would be great to update the status in Homekit

wayn3, 2020-06-19:
👆🏼

HonzaJaros, 2020-07-01:
Anything is happening in this topic, I have here sonoff RFR3 controlled by HAA and little people control it by RF. So would be great if this is implemented

wayn3, 2020-07-12:
label "wontfix" = 😞

tonysprenk, 2020-07-13:
Yes but it is understandable. Perhaps someone with more coding knowledge than me can create a PR or a fork... I am actually starting to learn C but I'm afraid it will take a while for me to be proficient enough...

wayn3, 2020-07-13:
@RavenSystem would you merge in stable pull requests from forks if someone was to take this on?

RavenSystem, 2020-07-21:
> @RavenSystem would you merge in stable pull requests from forks if someone was to take this on?

Yes, but must be a good work, not only supporting specified devices. And it must be RAM memory friendly; this is the hardest part.

tonysprenk, 2020-08-16:
.

tonysprenk, 2020-09-10:
.

hillaliy, 2020-09-14:
Tasmota support receiving UART commands. Maybe you can copy somthing from the code.

tonysprenk, 2020-09-14:
AFAIK Tasmota is not written in C.

tonysprenk, 2020-11-02:
.

guatedude2, 2020-12-17:
Looks like Tasmota's TuyaMCU driver is written in C
https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_16_tuyamcu.ino this could be promising to enable a way to add TuyaMCU support.

paullj1, 2021-01-18:
Hey all! I’m looking at trying to implement this, and want a place to write my thoughts, and hopefully sanity check it.  So I see a few different ways of doing this. We could implement a raw/generic UART message definition to watch in the config (the way the UART message sending works), or, implement the actually Tuya MCU and map dpid, and fnid the way that Tasmota does it. Either way, I think you’d want to use Tasmota to enumerate which dpids map to which functions, then flash HAA.

I’m leaning toward the Tuya protocol, but have very little embedded systems programming  experience.  So, I think I need to implement an interrupt handler that parses the messages that come back from the MCU, and then notify HomeKit, right?

For my particular case (#1259) I only care about the heartbeat, when the power goes off (dpid 1 value 1 or 0), and dimmer (dpid 2, value 10-1000). So the config would map dpid to a function id.

Does all of that make sense?  Thoughts anyone?

ericyanush, 2021-01-30:
I was looking to try implementing this as well once work calms down in a few weeks, but I’m happy to see someone else is interested in implementing this! 

I personally think that implementing the Tuya protocol is the best way to accomplish support of the tuya devices. While I’m sure generic uart messages could be configured for tuya devices,  it would likely be quite error prone for users to setup.

paullj1, 2021-03-06:
So the tuya protocol isn’t too hard to implement. The guys at Tasmota put together a great little driver for it. I simplified it in my project here: https://github.com/paullj1/TreatLife-DS02S-HomeKit/tree/main

That said, it seems the owner of this repo is unwilling to integrate the support. I think the most ideal way forward would be to fork this repo, strip it down to the most minimal functionality, and then implement only a JSON configurable Tuya device. Otherwise, you’ll be battling @RavenSystem for size in the flash memory.

tonysprenk, 2021-03-06:
@paullj1 Would you be willing to work out the solution you mention? My coding skills are none existent but I do have some curtains that would benefit greatly from this.

paullj1, 2021-03-07:
@tonysprenk I don’t have a particular use for it anymore.  It would be far easier to adapt the project I made for your curtains.  The main advantage of HAA is for more complex configurations, but in the case if the Tuya MCU, most of the inputs are handled directly by the MCU, so you can’t override them easily to extend their functionality. In other words, I’m not sure what you gain by making HAA handle Tuya, versus just modifying the base code I have linked in my project for your application.

peros550, 2021-03-10:
@paullj1 Reason why everyone is so obsessed with HAA is its incredible stability and improvements over time. 

Taking into account the flash storage constraints, it looks any special new features outside @RavenSystem radar have no luck. 
 
However, before someone would try to strip down HAA, it would make sense to ask @RavenSystem if he would be willing/interested/have time to support such approach?

In this way, any improvements or security fixes on HAA's baseline code by @RavenSystem would easily be adopted to any other firmware which was based on HAA's striped down code. I hope this make sense.

paullj1, 2021-03-12:
Any interest in a larger image for devices with 2mb flash (ESP8266 EX)?

closed 2021-08-22 by RavenSystem