RavenSystem/esp-homekit-devices #916: UART commands

asbesto01, 2020-05-01:
I have seen the demo of UART Commands running on the Shelly dimmer. It seems an interesting feature, can it be applied also to sonoff basic, for example, to manage lights dimming?

Gracias!

shanghaix, 2020-05-04:
That's not possible as the Sonoff basic lacks the internal hardware to be able to dim the lights. It's like asking if a Fiat can do the same things as a Ferrari.

asbesto01, 2020-05-04:
> That's not possible as the Sonoff basic lacks the internal hardware to be able to dim the lights. It's like asking if a Fiat can do the same things as a Ferrari.

Super clear!

Any device in the sonoff range compatible with uart dimming?

Otherwise I should use the Shelly dimmer!

shanghaix, 2020-05-04:
The Sonoff D1 (dimmer) came out earlier this year but so far it's only been compatible with Tasmota https://github.com/arendst/Tasmota/issues/7598. If I were you I would buy a shelly, they have a model that runs without a neutral wire coming out this month, on top of the regular one, and they are smaller than the Sonoff D1.

asbesto01, 2020-05-04:
> The Sonoff D1 (dimmer) came out earlier this year but so far it's only been compatible with Tasmota https://github.com/arendst/Tasmota/issues/7598. If I were you I would buy a shelly, they have a [...]

Let me give you a little bit of background:

Currently I have dimming capable lights, connected to sonoff basic with haa flashed, and physical buttons soldered to them to manually turn on and off lights.

Working perfect, but I want to add dimming. I have neutral everywhere (currently using it with the sonoff basic).

This considered, which is my best option? I want HAA, I want dimming, I have neutral everywhere.

Which is the perfect Shelly for me?

Thanks a lot!

shanghaix, 2020-05-05:
Then I would go with the regular shelly dimmer, its the best bang for your buck at the moment if you want a dimmer and use HAA to get homekit compatiblity!

asbesto01, 2020-05-06:
> Then I would go with the regular shelly dimmer, its the best bang for your buck at the moment if you want a dimmer and use HAA to get homekit compatiblity!

Great, can someone please share a config.json for the Shelly dimmer supporting uart dimming via HomeKit haa?

Thank you so much!

shanghaix, 2020-05-07:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database

asbesto01, 2020-05-08:
> Config is there:
> https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database

ok purchased one; I see that for programming I need to use this wiring scheme:

[image]

the only problem is that I don't know which pins to short-cut to enter flash mode. Any suggestions?

tonysprenk, 2020-05-08:
Gpio0 to gnd during startup.

asbesto01, 2020-05-13:
Great, I correctly flashed it.

Now I have an issue; the shelly is able to actually dim my dimmable lamps, however at 0% it does not shut off, only flickering  (and intense) light.

What am I doing wrong?

RavenSystem, 2020-05-20:
Shelly Dimmer is not a good device, and it has some incompatibility with some lamps. I used them with Amazon lightbulbs and results were bad. Better use a TRIAC based dimmer.

closed 2020-05-20 by RavenSystem

asbesto01, 2020-05-20:
Any suggestions on HAA dimmers?

tokamac, 2020-10-16:
> > any suggestions on TRIAC dimmers that can be flashed with HAA?

Have a look at **QS-WiFi-D01-TRIAC**, it is listed in the [device database](https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database) also. Please report here your findings, trials and errors with this one.

asbesto01, 2020-12-19:
> QS-WiFi-D01-TRIAC

> > > any suggestions on TRIAC dimmers that can be flashed with HAA?
> 
> Have a look at **QS-WiFi-D01-TRIAC**, it is listed in the [device database](https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database) also. Please report here your findings, trials and errors with this one.

Thank you! Do you have a guide to flash it? And also the correct json please!