RavenSystem/esp-homekit-devices #917: NodeMCU won't leave Setup mode / won't start HomeKit

vsaintloup, 2020-05-01:
Hello, I'm quite new at this, I'm pretty sure I followed all the steps carefully… any help would be greatly appreciated!

I used a very simple existing JSON example just to test it out:
`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"b":[{"g":0}],"s":0,"p":1024,"d":1,"e":10,"fr":1,"fg":1,"fv":1}]}`

Saving seems to be successful since after rebooting I see my JSON code…
Here's what I get after rebooting the unit: 

```
pm_task_hdl : 3ffef830, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3f78, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr 30 2020 19:26:50
phy ver: 273, pp ver: 8.3

Sysparam ready
mode : sta(50:02:91:dc:d9:85)
add if0
Fatal exception (28): 
epc1=0x4000bf80
epc2=0x00000000
epc3=0x00000000
excvaddr=0x00000000
depc=0x00000000
excsave1=0x40231c10
Registers:
a0 40231c10 a1 3fff4e10 a2  00000000 a3  fffffffc
a4  000000ff a5  0000ff00 a6  00ff0000 a7  ff000000
a8  00000000 a9  00000000 a10 00000003 a11 7fdbc255
a12 3fff433c a13 3fff5840 SAR 0000001f

Stack: SP=0x3fff4e10
0x3fff4e10: 3fff648c 3fff5840 40249825 40231bee
0x3fff4e20: 3fff4f50 00000001 4024483b 40217aec
0x3fff4e30: 3fff4f50 00000001 4024483b 40218b23
0x3fff4e40: 0000000c 3fff433c 3fff5b98 40226ef3
0x3fff4e50: 34000031 a5a5a500 a5a5a5a5 40105d9a
0x3fff4e60: 3fff42f0 00000001 3fff08e8 40105b9c
0x3fff4e70: 3fff5050 3fff4156 3fff5870 40105aca
0x3fff4e80: a5a5a5a5 a5a5a5a5 a5a5a5a5 402262c5

Free Heap: 39748
_heap_start 0x3fff1df8 brk 0x3fff64bc supervisor sp 0x40000000 sp-brk 39748 bytes
arena (total_size) 18116 fordblks (free_size) 0 uordblocks (used_size) 18116

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff40, prio:14, stack:512
pm_task_hdl : 3ffef830, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3f78, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr 30 2020 19:26:50
phy ver: 273, pp ver: 8.3

Sysparam ready

Home Accessory Architect v2.1.1
Developed by José Antonio Jiménez Campos (@RavenSystem)

SETUP MODE
WiFi config init
WiFi Mode: Normal
HAA Setup
Enabling auto reboot
Start AP SSID=HAA-DCD985
Start DHCP server
mode : sta(50:02:91:dc:d9:85) + softAP(52:02:91:dc:d9:85)
add if0
add if1
bcn 100
Start WiFi scan
scandone
Start DNS server
Start HTTP server
Hostname: HAA-DCD985
scandone
reconnect
scandone
add 0
aid 22
cnt 

connected with AirPort Extreme 2.4 GHz, channel 6
dhcp client start...
ip:10.0.1.53,mask:255.255.255.0,gw:10.0.1.1
bcn 0
del if1
mode : sta(50:02:91:dc:d9:85)
Stop DNS server
scandone
scandone
scandone
scandone
scandone
scandone
```

It stays in Setup mode forever, no matter what JSON string I use, and never goes discoverable in HomeKit. 

What am I missing?
Thanks so much!

RavenSystem, 2020-05-01:
Remove `"p":1024` from JSON. Maybe other things are wrong. Web configurator has many bugs.

vsaintloup, 2020-05-01:
@RavenSystem , Wow you got it! It works! (and it works beautifully, great job!) 
Thanks!

closed 2020-05-01 by vsaintloup